bigtreetech/BIGTREETECH-SKR-V1.3 #387: [BUGS] SKR 1.4 Turbo + TFT35 V3.0 + TMC2209s

skappers, 2020-08-01:
Hi all, 

I have compiled Marlin 2.0.5 onto the above and am experiencing multiple issues which seem to be common when I look through Github but none of the suggested fixes have worked for me. I have compiled the firmware to not use sensorless homing. 

1. Printer not found error in touch screen mode. I have adjusted the baurate multiple times with no success.

2. Can't find board when connecting to computer via USB. I can flash firmware via SD card and have set up the ports correctly but still can't find the board or communicate to it via Pronterface. 

3. When switching to Marlin Mode I get a MAXTEMP: E1 error. I have read people who have had this issue have been able to fix it compiling older versions of Marlin which I haven't tested yet. I have run a multimeter across the thermistor sockets on the board with power turned off and thermistor plugs for bed and hot-end not connected. E0 reads as closed circuit while E1 and the bed are both open. 

Hoping the community is able to assist - I have attached both my config and config.adv files. 

[Archive.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/5009658/Archive.zip)

edwilliams16, 2020-08-01:
re 1) I didn't get as far as hooking up the touch-screen before my boards failed.

re 2) When I flash off-line and put in the SD card in the SKR and boot, it does change FIRMWARE.BIN to FIRMWARE.CUR  However, I can't see the board via USB.

re 3)  I just checked, for me all three thermistor plugs register as open circuit - which the schematic suggests they should.

edwilliams16, 2020-08-01:
https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/issues/223  addresses the MAXTEMP, though was inconclusive.

skappers, 2020-08-01:
Thanks @edwilliams16, I took a look at the thermistor at the hot end and the glass looks burnt out to me which would be causing the short (hopefully), waiting for replacements to arrive to confirm. Regarding the screen issue, I think the boards faulty. Waiting to hear from BigTreeTech

edwilliams16, 2020-08-01:
Does
#define TEMP_SENSOR_0 998   // dummy thermistor 25C

solve any of your other problems - in particular the lack of a USB connection?

skappers, 2020-08-02:
@edwilliams16  this confirmed the thermistor is short. I did your suggestion and the usb communicates to the computer - I have been able to connect the board via Pronterface also. 

I still get No Printer attached error in touch screen mode but I am going to try changing baurate back to 25000. 

Marlin mode is fine - I do get a TMC Connection error but only when board is powered by USB.

edwilliams16, 2020-08-02:
I re-measured the resistance between the TH0 pins with the power off and the thermistor disconnected - I get a short.  Measuring the thermistor resistance through the cable I get 100K.  With TH1 and TB I get 6.2K.  So the board is shorted not the thermistor.  This is the same for my other dead board.
I connected up a reprep-discount-smart-graphics-controller and the board boots up giving a MAXTEMP error.  This went away when I did the TEMP_SENSOR_0 998 trick. HOWEVER, in my case when I attempt to connect through the USB to the board, my Mac crashes with a kernel panic - no Pronterface for me.
LATER:  If I connect the Mac up to the USB before I power the board, it CONNECTS!  
So at this point everything so far works except the extruder thermistor, which is a fault on the board.  I might be able to work around  by swapping TH0 and TH1 in the pins file. @skappers

skappers, 2020-08-02:
@edwilliams16 sure sounds like it! I haven't tried swapping the pins in the pins file, will give that a go next. Testing the TH0 pins with power off I also get a short even though the thermistor is disconnected. I have raised a dispute with Bigtreetech and provide them with evidence of my tests, I will let you know how I go with them.

AnHardt, 2020-08-02:
> changing baurate back to 25000.

Don't you mean 250000?

A short can often be found by visual inspection. Or just removing from the housing.

edwilliams16, 2020-08-02:
"Just removing from the housing"?  I'm not sure what you are referring to - the JST connector?  I would think the most likely place for a short at TH0 would be failure of RV10 or possibly C18. (see above) The CPU should be OK, I think, as the 4.7K series resistor would limit the current to 5mA and they are rated for 25mA, even if you somehow managed to apply 24V.  However, I'm a physicist not an electronics guy, so please set me straight if I'm on the wrong track.

AnHardt, 2020-08-02:
I'm referring to vagabonding metal parts like solder chips, wire strands, ... under the board, between the T-contacts, between processor legs, ... .

edwilliams16, 2020-08-03:
OK. I understand. The board looks clean and no visible damage.

edwilliams16, 2020-08-03:
@skappers  Re swapping TH0 and TH1 in the pins file.  You might expect to swap them in PINS_BTT_SKR_common.h since they appear there.  
// Temperature Sensors
//  3.3V max when defined as an analog input
//
#ifndef TEMP_0_PIN
  #define TEMP_0_PIN                    P0_24_A1  // A1 (T1) - (68) - TEMP_0_PIN
#endif
#ifndef TEMP_1_PIN
  #define TEMP_1_PIN                    P0_25_A2  // A2 (T2) - (69) - TEMP_1_PIN
#endif
#ifndef TEMP_BED_PIN
  #define TEMP_BED_PIN                  P0_23_A0  // A0 (T0) - (67) - TEMP_BED_PIN
#endif
but in fact they shouldn't really be there as 1.3 and 1.4 boards have different pin numbering. They are over-ridden for 1.4 (turbo) in
PINS_BTT_SKR_1.4.h where I changed

#define TEMP_1_PIN                      P0_23_A0  // A2 (T2) - (69) - TEMP_1_PIN
#define TEMP_BED_PIN                    P0_25_A2  // A0 (T0) - (67) - TEMP_BED_PIN

to

// swapped 1 and 0
#define TEMP_0_PIN                      P0_23_A0  
#define TEMP_1_PIN                      P0_24_A1
#define TEMP_BED_PIN                 P0_25_A2

I'm actually up and running now.  Hopefully the TH1 port will hold together...  My Mac still crashes 50% of the time when I connect to the board.  <sigh>

skappers, 2020-08-07:
glad to hear you could get it up and running @edwilliams16, I will have a go at swapping the pins around. Working through trouble shooting with Bigtreetech to confirm the short and get a replacement board sent out (have to provide video evidence).

closed 2020-08-17 by skappers

MohanBajoju, 2023-08-03:
Will it possible to control the PWM FAN0 with TH0  thermistor using btt skr 1.4  motion controller board.