bigtreetech/BIGTREETECH-SKR-V1.3 #388: Thermistors T0 and T1 didn't work

Jon-Oak, 2020-08-02:
My SKR 1.3 was working fine when the thermistor wire accidentally touched against aluminum block (I didn't see the wire was out of the plastic tube) and T0 shows 40ºC, even if nothing connected to the board. I've tried to change T0 to T1, but it  didn't work, it shows 40ºC. 
The board have any other port to change and use for hotend thermistor?

edwilliams16, 2020-08-02:
The last three posts are all about failed thermistor ports.   But I'm puzzled because your aluminum (hotend I assume) block should, if anything, be at ground potential and grounding the signal pin on the thermistor port shouldn't break the board - it ought to  just give a MAX_TEMP error and shut it down. If it does, that would explain why there are so many problems as this is clearly an accident waiting to happen.
  OTOH, if somehow your hot-end heating element has lost its insulation and the block is at supply voltage (12V/24V), I'm not at all surprised if that would damage the port.  The TVS diodes can only handle transient voltage spikes. Can you put a voltmeter on your block and see what is happening? While you are at it, measure the resistance (board unpowered!) across the T0 and T1 pins to see if they have developed an internal short.

edwilliams16, 2020-08-03:
I just swapped from TH0 to TH1 on my 1.4_turbo.  It worked fine.  In your case you need to swap the definitions of
TEMP_0_PIN and TEMP_1_PIN  in in PINS_BTT_SKR_common.h
 Is that what you did?

Jon-Oak, 2020-08-06:
I will verify with voltmeter. 

I tried to change in pins_RAMPS.h, but it didn't work. This archive PINS_BTT_SKR_common.h I will swap TEMP_0_PIN definitions to TEMP_1_PIN, like this:
Before swap:
#ifndef TEMP_0_PIN
  #define TEMP_0_PIN       P0_24_A1   // A1 (T1) - (68) - TEMP_0_PIN
#endif
#ifndef TEMP_1_PIN
  #define TEMP_1_PIN       P0_25_A2   // A2 (T2) - (69) - TEMP_1_PIN
After swap:
#ifndef TEMP_0_PIN
  #define TEMP_0_PIN       P0_25_A2   // A2 (T2) - (69) - TEMP_1_PIN
#endif
#ifndef TEMP_1_PIN
  #define TEMP_1_PIN       P0_24_A1   // A1 (T1) - (68) - TEMP_0_PIN

edwilliams16, 2020-08-06:
That should work for your 1.3 board. In the 1.4 case, you need to make the changes in PINS_BTT_SKR_1.4.h

skappers, 2020-08-18:
I had the similar issue and THO was actually shorting at the board, I got a refund from Bigtreetech but changed the pins in Marlin as @edwilliams16  has suggested and that resolved my issue.

Jon-Oak, 2020-09-30:
> Acabei de trocar de TH0 para TH1 no meu 1.4_turbo. Funcionou bem. No seu caso, você precisa trocar as definições de
> TEMP_0_PIN e TEMP_1_PIN em PINS_BTT_SKR_common.h Foi
> isso que você fez?

Hello, sorry for the long time to answer. I've changed  like you said and it works perfectly.