bigtreetech/BIGTREETECH-SKR-V1.3 #389: CR10-S Display not working using both exp ports

Spidersky1489, 2020-08-03:
I have compiled my firmware to use the RepRapDiscount Full Graphic Smart Controller (which is the display that comes with the CR10-S), however when I connect the display to the board with the SD card with the firmware on that SD card, the display backlight does not turn on but the firmware has flashed, as I am able to control it via Octoprint, meaning that the printer firmware is working correctly except for the display. When I was using both EXP1 and EXP2 connectors, I would make movements using octoprint and after a few seconds I would hear the buzzer make a faint and delayed sound. This confused me as I thought that the connectors simply did not work. Since this version of the display had the EXP3 connector used by the original CR10 display, I compiled the firmware to use that display and it now works somehow. I also hear a long beep when the printer turns on and I wanted to ask if there was a way to disable that.

I have attached my configuration.h and configuration_adv.h for anyone to take a look and diagnose if they want to.

Thanks for any help.

[Marlin.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/5018969/Marlin.zip)

JRosello, 2020-08-03:
Try to connect the display on EXP1 and EXP2 reversing the connectors, you must clip the notch on each cable.
I need to do this when I upgraded to SKR 1.4T.

Spidersky1489, 2020-08-04:
> Try to connect the display on EXP1 and EXP2 reversing the connectors, you must clip the notch on each cable.
> I need to do this when I upgraded to SKR 1.4T.

To my surprise it works now! Thank you very much, thought I tried this earlier but I guess not!

closed 2020-08-04 by Spidersky1489