bigtreetech/BIGTREETECH-SKR-V1.3 #390: [GUIDE] SKR 1.4 TURBO with LJ18A3-8-Z/BX

gec100, 2020-08-04:
Had this sensor for long time working great on my custom made prusa clone and now I'm building a new corexy printer around SKR 1.4 Turbo board on 24V.

I'm not a electronic engineer but I can find my way around. The sensor needs 6-36V to work properly. On 12V boards this sensor needed to be connected to the 12V pin (spare Fan connector) with voltage divider http://reprap.com/wiki/Z_probe

this being 24V setup, resistors had to be changed. with https://www.allaboutcircuits.com/tools/voltage-divider-calculator/
I calculated that my resistors need to be roughly 1:4 in size to get the output voltage below 5V needed for the Z-stop probe pins. That would be cca 4kΩ for R1 and 1kΩ for R2 or 8kΩ for R1 and 2kΩ for R2 . You get the point. I had at my hands 5k1 for R1 and 1k for R2 giving me cca 3.9V and it worked.

here is my very rough schematics: 
[image]

As I said I'm not an electronic engineer but this works for me. If you have any comment I would be grateful.

lautarobusto, 2020-12-09:
thank you very much. saw a buch of vids on how to use this probe but all used  ardui+ramps  and similiar  probes using 5v, but no one mentioned  something about using a voltage divider to get the signal to the correct voltage.

gec100, 2020-12-10:
glad to help when I can