bigtreetech/BIGTREETECH-SKR-V1.3 #393: BuildingSKR-1.4 Turbo Firmware, completely stock, HAL error

Vnldwth, 2020-08-09:
Afternoon all, I'm trying to build and use the skr 1.4 turbo, replacing my fried 1.3 with it.  
I configured everything, including the UART settings for the 2208's, and I'm getting a rather weird error?  I've tried a few different things, but can't get to to finish.  any ideas?  log below.

`> Executing task in folder Marlin-2.0.x-SKR-V1.4-Turbo: C:\Users\evanl\.platformio\penv\Scripts\platformio.exe run <

Processing LPC1769 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.2.zip; board: nxp_lpc1769; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nxplpc-arduino-lpc176x/nxp_lpc1769.html
PLATFORM: NXP Arduino LPC176x 0.1.1 > NXP LPC1769
HARDWARE: LPC1769 120MHz, 31.97KB RAM, 464KB Flash
DEBUG: Current (cmsis-dap) On-board (cmsis-dap) External (blackmagic, jlink)
PACKAGES:
 - framework-arduino-lpc176x 0.2.4
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ strict
Found 6 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Servo> 1.0.0
|-- <LiquidCrystal> 1.0.0
|-- <U8glib-HAL> 0.4.1
|-- <TMCStepper> 0.7.1
|-- <Adafruit NeoPixel> 1.3.0
|-- <SailfishLCD>
Building in release mode
upload disk:  G:
Compiling .pio\build\LPC1769\src\src\HAL\HAL_LPC1768\DebugMonitor.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\HAL_LPC1768\HAL.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\HAL_LPC1768\HAL_SPI.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\HAL_LPC1768\MarlinSerial.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\HAL_LPC1768\fast_pwm.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\HAL_LPC1768\include\digipot_mcp4451_I2C_routines.c.o
Compiling .pio\build\LPC1769\src\src\HAL\HAL_LPC1768\main.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\HAL_LPC1768\persistent_store_flash.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\HAL_LPC1768\persistent_store_sdcard.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\HAL_LPC1768\timers.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\HAL_LPC1768\u8g\LCD_I2C_routines.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\HAL_LPC1768\u8g\LCD_pin_routines.c.o
Compiling .pio\build\LPC1769\src\src\HAL\HAL_LPC1768\u8g\u8g_com_HAL_LPC1768_hw_spi.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\HAL_LPC1768\u8g\u8g_com_HAL_LPC1768_ssd_hw_i2c.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\HAL_LPC1768\u8g\u8g_com_HAL_LPC1768_st7920_hw_spi.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\HAL_LPC1768\u8g\u8g_com_HAL_LPC1768_st7920_sw_spi.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\HAL_LPC1768\u8g\u8g_com_HAL_LPC1768_sw_spi.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\HAL_LPC1768\usb_serial.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\HAL_LPC1768\watchdog.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\shared\HAL_spi_L6470.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\shared\backtrace\backtrace.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\shared\backtrace\unwarm.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\shared\backtrace\unwarm_arm.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\shared\backtrace\unwarm_thumb.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\shared\backtrace\unwarmbytab.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\shared\backtrace\unwarmmem.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\shared\backtrace\unwinder.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\shared\backtrace\unwmemaccess.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\shared\eeprom_i2c.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\shared\eeprom_spi.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\shared\persistent_store_api.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\shared\servo.cpp.o
Compiling .pio\build\LPC1769\src\src\Marlin.cpp.o
Compiling .pio\build\LPC1769\src\src\core\multi_language.cpp.o
Compiling .pio\build\LPC1769\src\src\core\serial.cpp.o
Compiling .pio\build\LPC1769\src\src\core\utility.cpp.o
Compiling .pio\build\LPC1769\src\src\feature\I2CPositionEncoder.cpp.o
Compiling .pio\build\LPC1769\src\src\feature\Max7219_Debug_LEDs.cpp.o
Compiling .pio\build\LPC1769\src\src\feature\babystep.cpp.o
Compiling .pio\build\LPC1769\src\src\feature\backlash.cpp.o
[... 269 lines cut ...]
ARMED                         IGNORED
STM32F103VE_longer            IGNORED
mks_robin                     IGNORED
mks_robin_pro                 IGNORED
mks_robin_lite                IGNORED
mks_robin_lite3               IGNORED
mks_robin_mini                IGNORED
mks_robin_nano                IGNORED
jgaurora_a5s_a1               IGNORED
STM32F103CB_malyan            IGNORED
chitu_f103                    IGNORED
FLYF407ZG                     IGNORED
FYSETC_S6                     IGNORED
STM32F407VE_black             IGNORED
BIGTREE_SKR_PRO               IGNORED
BIGTREE_BTT002                IGNORED
teensy31                      IGNORED
teensy35                      IGNORED
esp32                         IGNORED
linux_native                  IGNORED
SAMD51_grandcentral_m4        IGNORED
rumba32_f446ve                IGNORED
mks_rumba32                   IGNORED
include_tree                  IGNORED
========================================================================================================================================== 1 failed, 0 succeeded in 00:00:47.133 ================================================`

edwilliams16, 2020-08-10:
Why mess with a 7 month old version of Marlin.  Use a current one from the Marlin repository.

Vnldwth, 2020-08-10:
> Why mess with a 7 month old version of Marlin. Use a current one from the Marlin repository.

Realized I had a few things wrong...  ended up wiping all my saved Marlin copies, either stock or distributed, and starting over.  an extra 30 minutes of sifting through config and machine settings and it worked.  Thanks for replying!

closed 2020-08-10 by Vnldwth