bigtreetech/BIGTREETECH-SKR-V1.3 #394: SKR 1.4 Turbo + TMC2209 drivers + Motor movement problem

WestendVince, 2020-08-10:
Hello, I've updated a Mamorubot SX4 core xy printer with an SKR 1.4 turbo and TMC2209 drivers. The z axis is working correctly after increasing the current from 800mA to 1100mA, smooth and quiet as advertised. The x and y axis are moving in the right directions, though loudly, if I call for 1mm or .1mm from the Pronterface app.  If 10mm is called for, the axis howls or buzzes while moving .5mm in the correct direction, then moving back.
Attached are the current config files and output of the m122 command.  Any help is appreciated.  Since my stroke my troubleshooting skills are severely diminished but I follow direction well.

[m122Output.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/5053078/m122Output.txt)
[Configuration-h.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/5053092/Configuration-h.txt)
[Configuration-adv.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/5053093/Configuration-adv.txt)

.

Arakon, 2020-08-17:
That usually means you are trying to make the motors move faster than they can physically move. What happens if you reduce the feed rate/movement speed?

WestendVince, 2020-08-19:
Thanks for your reply. I've reduced the acceleration and now x&y are better.  My current hurdle is the bl-touch disconnects the board from the laptop when I connect the black and white probe wires.  I've tried them in the labeled connector and in the z end stop location with the same results. Vince Plouzek 001 406 274 1456..Sent from my Samsung Galaxy smartphone.

  drivers + Motor movement problem (#394) 

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/issues/394#issuecomment-674739006",
"url": "https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/issues/394#issuecomment-674739006",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

Arganaz, 2020-08-29:
-Pronterface has its own "speed" settings (and most slicer software), check on the right side of the buton "Motors off" if its the default 3000, try to change to 500, and see if it helps.

- for the noise, try checking your HYBRID_THRESHOLD settings. low info but, stealthChop is quiet and spreadCycle is powerfull and louder.

WestendVince, 2020-08-29:
Thank you for your assistance, I'll check that out.Vince Plouzek 001 406 274 1456..Sent from my Samsung Galaxy smartphone.

WestendVince, 2020-09-29:
thank you all for the help. motor movement has stabilized to the point I've started on homing now.

closed 2020-09-29 by WestendVince