bigtreetech/BIGTREETECH-SKR-V1.3 #395: SKR v1.4 turbo USB Communication issue

armature81, 2020-08-14:
I just got new SKR1.4TURBO
i was able to connect on time only then i did change in Marlin confi.h though ou VScode
then I cannot connect via my computer 
- On my computer, it doesn't show up in Device Manager or make a noise when it gets connected.
- The firmware.bin doesn't change to a .cur and the default .cur file alone doesn't load anything either
- I have tried multiple sd cards, none have had any different a result
- The SD indicator LED turns on meaning the board detects the SD card's existence

Arganaz, 2020-08-29:
> * On my computer, it doesn't show up in Device Manager or make a noise when it gets connected.
You should at least see a serial port for your printer (ie: COM4)
> * The firmware.bin doesn't change to a .cur and the default .cur file alone doesn't load anything either
delete(or move) the cur file the bin is the one you need (you can also move/rename it, to make sure your producing a new one from VSCode)
> * I have tried multiple sd cards, none have had any different a resulttry changing to _#define SERIAL_PORT 0_ and #define SERIAL_PORT_2 -1 and see if it helps (allow some time for windows to assume it after printer reboot)
> * The SD indicator LED turns on meaning the board detects the SD card's existence
Most videos/pages use their own configs, try to change it so it matches yours. I had a similar prob and this solved it for me.

hope it helps

soperskaft, 2020-11-07:
So. did it work?