bigtreetech/BIGTREETECH-SKR-V1.3 #396: SKR 1.4 turbo TMC 2209 drivers firmware not updating

bobh26501, 2020-08-15:
Firmware will not update. Have tried different micro sd cards, adapters. Powered board from printer and USB. Have tried with board in and out of printer. Powered from computer and usb pwr supply. With and without driver installed. When connected via usb to computer it shows up as marlin device in windows 10. Have a BBT Writer. Can this be used to install bootloader/firmware to board? Other options for updating firmware? Or is the board bad?

wilsongis, 2020-08-16:
I am having the same issue. My BTT SKR 1.4 Turbo cannot see the sd card. I can't update the firmware.

Hyrules, 2020-08-20:
Same here. I suspect a bad firmware. the version on the board is 2.0.0.5 github. I have ordered a STLInk to flash the firmware using the serial like I saw people do it that way. Supposedly the firmware is not configured properly.

wilsongis, 2020-08-20:
I went ahead and exchanged the board. It is working now.

Hyrules, 2020-08-20:
Same here. Good thing I purchased the board with amazon prime. Return no question asked. Will get a new one tomorrow.

wilsongis, 2020-08-20:
> Same here. Good thing I purchased the board with amazon prime. Return no question asked. Will get a new one tomorrow.

Seems like it is a real quality issue though. My new board is working although I am still tweaking. Hope your works as well.

Hyrules, 2020-08-24:
New board is working fine now for me as well.

pumi1963, 2020-08-27:
USB driver:  Marlin-> smoothieware...

mvinkarklins, 2020-08-30:
Hello, guys have exactly the same problem with SRK 1.4 turbo - could you send me a link to the video on how to use ST-LINK  fot bootloader flash, please?

Hyrules, 2020-08-30:
Don't have the exact procedure but this thread seems to give a good idea : https://community.platformio.org/t/skr-1-4-bricked/12932/10

helgjohn, 2020-09-04:
SKR 1.4 turbo board Firmware will not update, no Sd card detection so no printing via SD card only via usb / raspberrypi any one got a fasit to make it work..... Marlin ver onboard is 2,0,5.4

docbobo, 2020-09-06:
Same problem here. Running Marlin 2.0.6 and  wanted to update to Marlin 2.0.6.1. Nothing happening...

wilsongis, 2020-09-07:
> Same problem here. Running Marlin 2.0.6 and wanted to update to Marlin 2.0.6.1. Nothing happening...

I had to buy a new board. Once I did that, I was good to go.

docbobo, 2020-09-07:
Ina somewhat reluctant to do that. Don’t want to replace boards every other weeks - that’s quite some work.

helgjohn, 2020-09-08:
> > Same problem here. Running Marlin 2.0.6 and wanted to update to Marlin 2.0.6.1. Nothing happening...
> 
> I had to buy a new board. Once I did that, I was good to go.

well that solved the problem for you, but its hardly a long term option that most owners of these cards want, granted they are cheap. But  just like most car owners dont buy a new car because the fuel cap is stuck they fix the problem with the cap.....

taarnby, 2021-01-04:
I have the same problem, thought it was maybe how I had formatet the sdcard, I have access to the card via usb to the sky 1.4 turbo board , but it doesn't take the bin file :( is there a way around this can it be done ?

julianCast, 2021-04-14:
The problem may be the crappy micro SD card reader built-in on our SKR 1.4.
When I plug the board with a USB, I can see the sd card BUT, I can barely do anything with it, extremely laggy and it takes like 10 minutes to be opened.

If I extract the sd card and use an external sd card reader no problem at all.

Arnie3d, 2022-02-06:
I was booting the whole time by PC and after a couple of bots it stoped booting also direct ond the board no luck

Arnie3d, 2022-02-06:
BTT come on dont tell me i bought another cat in the hat...just like the Octopus  with boot poort usb but wont boot

caiosms, 2023-07-23:
Guys, I had the same problem and the solve was put a strip of paper folded in two above the SD card when insert it. Aparently the socket have some clearance and the paper help to joint the card contacts.