bigtreetech/BIGTREETECH-SKR-V1.3 #397: SKR 1.4 Turbo: Heatbed out not works.

nesergen, 2020-08-15:
I bought a BIQU116-A2 mosfet module. Connected it to SKR 1.4 Turbo. It turned out that the mosfet does not work, the table heats up on its own immediately after turning on the printer. Then I turned on the table heating on the motherboard through the display menu and took a voltage measurement. It turned out to be close to zero. How can I check the functionality of the bed heating output on the SKR board? Why may the board lack output voltage?

nesergen, 2020-09-27:
I turned to the seller, at his request, I took measurements with a multimeter on the remote module of the mosfet. Based on the measurements, he claims that the heating mosfet on the motherboard is out of order. I checked the mosfet on the board with a tester. It opens and closes, changes its resistance, as if there were no damage. However, when the printer is turned on on the motherboard, the table heating output does not generate any voltage. What else on the board may fail so that the table will not heat up. Also, if you connect the table directly to the motherboard, then it does not heat up.

AnHardt, 2020-09-27:
Yellow 20A fuse?

nesergen, 2020-09-27:
> Yellow 20A fuse?

checked with a tester, ok

nesergen, 2020-09-29:
The board is ok, the voltage is giving out, but under certain circumstances, which I installed in the firmware. But the external mosfet turned out to be inoperative.

closed 2020-09-29 by nesergen

flow23, 2021-03-04:
Can we reopen this issue? I have the exact same problem.
@nesergen what was the underlying problem for you?
The MOSFET led turns off during printing and bed heating is turned off.
It happens randomly and can't be reproduced.

nesergen, 2021-03-04:
> Can we reopen this issue? I have the exact same problem.
> @nesergen what was the underlying problem for you?
> The MOSFET led turns off during printing and bed heating is turned off.
> It happens randomly and can't be reproduced.

I came to the conclusion that the beegtree mosfet module is defective. Communication with the seller did not give any results. I had another mosfet module, installed it and the printer works without problems. There was no time to repair the defective module.

flow23, 2021-03-04:
> I came to the conclusion that the beegtree mosfet module is defective. Communication with the seller did not give any results. I had another mosfet module, installed it and the printer works without [...]

So you changed the MOSFET Q2 (for reference https://3dwork.io/wp-content/uploads/2020/06/Esquema_electrico_SKR2.jpg)?
Where did you buy it?

nesergen, 2021-03-04:
> So you changed the MOSFET Q2 (for reference https://3dwork.io/wp-content/uploads/2020/06/Esquema_electrico_SKR2.jpg)?
> Where did you buy 

I get it, we are talking about different things. I had 2 problems. First, the board had a marlin firmware with incorrect settings, so the built-in mosfets did not work. Reflashing the printer with right settings helped resolve this issue. 
The second problem was in the external mosfet BIQU116-A2. I could not revive it, I just replaced it.

flow23, 2021-03-04:
> I get it, we are talking about different things. I had 2 problems. First, the board had a marlin firmware with incorrect settings, so the built-in mosfets did not work. Reflashing the printer with [...]

You are using the external mosfet for the heated bed then?

nesergen, 2021-03-04:
> 
> You are using the external mosfet for the heated bed then?

Yes, because the built-in mosfets can power a heated bed with a capacity of up to 144 watts, and my bed consumes about 200 watts. What is the capacity of your heated bed?

flow23, 2021-03-04:
> > You are using the external mosfet for the heated bed then?
> 
> Yes, because the built-in mosfets can power a heated bed with a capacity of up to 144 watts, and my bed consumes about 200 watts. What is the capacity of your heated bed?

If my math is correct it is exactly 144W (for reference https://shop.prusa3d.com/de/mk3mk3smk3s/198-magnetic-heatbed-mk52-24v-assembly.html?search_query=heatbed&results=24). It worked for a year without problems, printing mostly PET(G) with 80-90 degree Celsius. I guess the internal mosfet or the thermistor is faulty. Another guess would be that the power cables from the bed are bad. I disassembled everything and sometimes it's working and sometimes it's not. It's a pita.