bigtreetech/BIGTREETECH-SKR-V1.3 #399: SKR 1.4 Turbo SD Card not working

wilsongis, 2020-08-16:
The board will not recognize the SD card so I can update the firmware. I have tried multiple SD cards. I can connect to the board via Simplify3D. How can I either get the SD card working or flash without an SD?

edwilliams16, 2020-08-18:
Have you tried writing the FIRMWARE.BIN file to the SD Card while it is mounted directly to your PC/MAC and then  putting it in the SKR SD Card slot?  Reboot the SKR twice and put the card back in your PC and see if its name has changed to FIRMWARE.CUR.   If it has, then the SKR has been successfully flashed.

Eddiiie, 2020-08-19:
Format the SD card with FAT32?  Make sure it is less than 8GB in size?

Eddiiie, 2020-08-20:
I was just poking around looking for PWM pins, and lost access to the SD card, too.   
P0_07 is what made me lose it.  I assume P0_08, and P0_09 will do the same thing.   
Seems to me those should be protected pins.

Thank you for the suggestion @edwilliams16  !    Thought I bricked the board.

wilsongis, 2020-08-20:
> Format the SD card with FAT32? Make sure it is less than 8GB in size?

I tried it and no dice.

wilsongis, 2020-08-20:
> Have you tried writing the FIRMWARE.BIN file to the SD Card while it is mounted directly to your PC/MAC and then putting it in the SKR SD Card slot? Reboot the SKR twice and put the card back in [...]

Tried it with no luck.

wilsongis, 2020-08-20:
I went ahead and ordered a new board. It is working. I am thinking it is a quality control issue.

closed 2020-08-21 by wilsongis