bigtreetech/BIGTREETECH-SKR-V1.3 #400: SKR 1.4 Turbo - Dead Mosfet?

skappers, 2020-08-18:
Hi Github community, 

After fixing the short in the board with the TH0 terminal by swapping terminals THO and TH1 around in Marlin, I am now experience an issue with my hot end instantly hitting up as soon as I switch the power on. 

I have been doing some reading and this can happen if the thermistor is short or the heat cartridge is dead but I also saw that this could be because of the Mosfet's shorting as well. 

How do you go about testing Mosfet's for shorts? I have tested the voltage output for terminals for HB1 and HB0 respectively and neither is outputting 12V when I turn on the machine so I assume the issue is at the hot-end. 

Am I testing the Mosfet's correctly and if they are short, does anyone know where to get replacements for the skr 1.4 turbo board?

Thanks

lextc, 2020-09-03:
Same problem here. But mine is a little different. 

Last night I just finished setting everything up. extruder calibration, steppers calibration, PID tuning for both Bed and head, and started printing. After a while, I got a thermal runaway alert and the printer... Didn't stop. The head temp kept on rising, and I had to trun it off after 285C. When I tried to power it back on, the head temp keept on rising. I quit and left it overnight. Now, after trying again, the head dosen't heat but the bed is always on and heats by himself.

radek8, 2020-09-21:
If you don't dare solder a mosfet,
you can use an external Mofset and switch it with the pin that went to the original mofset.

https://www.aliexpress.com/item/32816837677.html?spm=a2g0s.9042311.0.0.27424c4deDO57A

lextc, 2020-09-22:
I ordered a new board, the mosfet was ok, the gate cip was fried.

DominicTWHV, 2026-08-24:
Bed mosfet on mine failed short randomly mid print and almost burned house down, seriously.