bigtreetech/BIGTREETECH-SKR-V1.3 #401: Which PINs have PWM?

Eddiiie, 2020-08-19:
Is there something on the schematic that shows which pins have PWM capability or do they all have PWM?
I have SKR 1.4 Turbo

Eddiiie, 2020-08-20:
I did some poking around and found that the following pins have PWM -
**Verified PWM works -
Z-Stop P1_27
PWRDET P1_00
I2CSCA1 P0_00
I2CSCL1 P0_01
SPI_SCK P0_07  (top middle) YES
SPI         P0_26  (closest to power and CPU) YES
NEOPIX  P0_24
PROBE   P0_10**

Did **not** test but expected to work:
SERVO   P2_02  Assumed works

Pins I tested but bad things happened (lost access to the on board SD card):
SPI_SCK  P0_07  (top middle)    DO NOT USE

Pins I did not test because bad things happened on related pins:
SPI_MISO P0_08  (top right)     DO NOT USE
SPI            P0_09  (bottom middle) DO NOT USE

Pins I expected to work but did not work for PWM or GPIO -
Maybe I need to change more in the config?
E1_Det P1_25
E0_Det P1_26
Y-Stop P1_28
X-Stop P1_29

Pins I did not test but may work:
E1 Stepper pins, 
E1 Enable  P1_16
E1 Step      P1_15
E1 Dir        P1_14

TFT Pins
TFT RX0     P0_03
TFT TX0     P0_02

Wifi Pins
Wifi TX3    P4_28
Wifi RX3    P4_29

yray-yym, 2020-10-06:
Why did you specify NEOPIX P0_24, isn't NEOPIX P1_24?

Eddiiie, 2020-10-12:
Fixed!  Thanks!

drom89, 2021-04-14:
Hello, all of them are 3,3V ?