bigtreetech/BIGTREETECH-SKR-V1.3 #403: NPWR (Neopixel) jumper with DC-DC still connected to on-board 5V

guru-florida, 2020-08-21:
This seems strange to me. Doing some continuity testing, if I remove the neopixel power jumper the left two pins (NPWR to on-board 5V) are still electrically connected. As shipped the jumper is installed to short these two pins, but with the DC-DC module mounted you'd switch the jumper to the VOT side to power the neopixel from the DC-DC converter. As the PCB is wired, moving the jumper to VOT side actually electrically connects all 3 pins of the NPWR jumper! This means the DC-DC converter gets wired to the on-board 5V regulator. I would think the NPWR jumper would be more like a switch, it's either in the on-board 5V position or the DC-DC VOT position.

You have a buck and a (I assume) linear regulator powering the same line. I suppose this could work as long as the voltages are equal, like two batteries in parallel, but doesnt seem desirable and in certain situations could cause a board failure. Since the buck uses the output as feedback for regulation it's likely to read the linear regulator output as it's feedback input as well (assuming no isolation). Meaning the buck will supply power first, then as the linear regulator drops voltage (+overheating) the buck will then take over. I haven't looked at the full schematics so correct me if I am wrong.

guru-florida, 2020-08-21:
WARNING!! Indeed. The two regulators are electrically connected! Not through the SKR board though, but through the BTT TFT35 LCD module. If you use both modes, so both the serial TFT and EXP1&2 wires are connected then the NPWR pin (on TFT serial header) and the 5V on the EXP headers are electrically connected TFT side. 

So choose whether to power the TFT module with NPWR or on-board 5V power and clip the pin on the other header. I see the on-board regular is in fact a buck, so since it is efficient and probably has plenty of power to spare I chose to clip the NPWR pin on the TFT serial cable (easily done).

Madbyte3d, 2021-11-12:
guru-florida. Interesting post. I noticed a periodic (every 2.5 min) 10C (+-5) temperature fluctuation on both bed and hotend probes. I narrowed down the fluctuation to the BTT TFT70 connected to my SKR1.4T. If I connect the serial TFT and EXP1 5v lines to an external 5v source the temperature fluctuation stops (down to +- 0.5C). I had this DCC module laying around (it came as a package with the board).  I installed it and the fluctuations remain. If I understand your post correctly, you recommend disconnecting only the NPWR line in the serial cable (easily done) and that's it?
[image]

guru-florida, 2021-11-12:
Yes, you got it. :+1:  Let me know if your issue goes away. I'm curious.

Madbyte3d, 2021-11-12:
I just disconnected the NPWR cable from the Serial TFT connector. I tried it with and without module, making sure I set the VOT jumper accordingly.  Fluctuations remain as in the screenshot. The only way the fluctuations go away are if I supply the TFT (serial and EXP1) with 5v from an external source (buck converter drawing 24V from the PSU stepped down to 5v). I guess is that it could be the TFT inducing noise in the SKR's 5V supply.  I tried two additional variations: 1. only powering the serial with external 5v  and 2. only powering the EXP1 with external 5v. In both cases the fluctuations were less but still present.  
I was hoping I could use the DCDC module to eliminate the need for an external 5v source but it seems that won't be the case.

Anyway, I'm not sure this is the right place to continue this topic. If it is not, I'll greatly appreciate If you can suggest another site like Discord to talk about this. Either way thanks for your help.

BTW the screenshot above is missing the scale which is 5 degrees C.  This is how the temperature graph looks like with both TFT connections drawing from an external 5v source. The scale is on the left.
[image]

guru-florida, 2021-11-12:
That's a strange issue. Sounds like you might need more capacitance on the power rail. You could try adding some high values like 1000uF or even a few 330uF would do. Try on the 5v rail first. On the 24v might be good too, but you have to find caps that have a 40v rating or higher which you might not have in your back pocket.

Madbyte3d, 2021-11-13:
This would be a more elegant/simple solution than using a buck converter. Where would connect the capacitor(s)? The thing is that I disconnected everything but the thermistors and the TFT70. With heater off (or on) the fluctuation remains.  Only when I disconnect the TFT it stops. How much can the TFT70 be drawing? So, capacitors across the both TFT connectors (serial and exp1 (5v and ground)?

guru-florida, 2021-11-13:
The capacitors are connected across the + and ground of power rails. During large changes in power draws they'll fill in while the source supply catches up to the change. (Mainly your house's power panel and the 120v : 24v transformer and any switching regulators.) You can place caps anywhere across the +/ground, like right at the screw terminals, and best near either the load that causes the power changes, or near the part of the system you are trying to protect. Either way, the caps will respond quickly and only the wiring itself will contribute to delay (small).

I'm thinking your issue is more than a power capacitance issue. I suspect  more of a wiring issue. Possibly ground loops. The thermocouples are very sensitive to ground noise/bounce since they produce only a small signal that is then amplified. Recheck wiring. Try to eliminate extra ground wires to the same system (which would form a redundant loop).  Ground wiring should be like a star. Also ensure that high current wires (like the heating bed) are not running near or next to the sensor wires which could pick up some of the current change using inductance. I'd really suspect inductance but you said it still does it with the bed off. Any other systems like the nozzle heater maybe? Is your noise pattern consistent with the nozzle heating cycle?

Madbyte3d, 2021-11-16:
Yes with everything off and everything disconnected except the thermistors and the TFT (serial, exp1, exp2). I managed to splice the EXP1 cable and connect the 5V and ground to the neopixel port with the DC-Dc module installed. This is what it looks like now.  Only concern is what you were talking about having both regulator connected