bigtreetech/BIGTREETECH-SKR-V1.3 #407: SKR V1.3 with TMC2209's Keeps Burning Up Extruder Motors

rrileypm, 2020-08-24:
I put together a CR-10 clone from scratch using an SKR V1.3 I ordered with TMC2209 drivers.

I now have everything configured to the point that I've started several test prints.  I've set this up with 2 extruders going into one shared nozzle.

I originally had current set at 800 for all stepper motors, including the two extruders.  I found that the default settings made the Y motor skip, but it works fine at 800.

During printing the E0 motor quit turning and started skipping back and forth.  It was too hot to touch, so I swapped over to the E1 motor, and it still worked for a couple of tests, then fried too.  These were little pancake Nema 17's that came with the extruders I bought.  I had a few extra larger Nema 17's laying around I've tried hooking up to see if they worked, and all I get is back and forth skipping motions instead of the shaft turning.  Have I also fried the stepper drivers?  In setup, I removed all of the jumpers from underneath the drivers, and just placed the ones on the board to activate UART mode.  (as instructed on the BigTreeTech setup video).

I now realize 800 might be too high for current.  I've reduced this number to 550 for X, Y and Z and 650 for both extruders.  I'm waiting on 2 new motors to arrive to replace the two motors I seem to have fried.

Is there a setting in the firmware I'm missing to keep this from happening?  What should be the normal settings for extruder motors with TMC2209's?  The new ones I ordered are 42x40mm Body Bipolar 1.5A 400mN.m 2 Phase which should be stout enough for the job.  Would it be necessary for me also replace the TMC2209's for both extruders?  I've put together several 3D printers, and I'm pretty good at troubleshooting, but this issue is tough for me to figure out.

I saw a thread about overheating problems with V1.0 and V1.2, but the solution appeared to be solved by an update in Marlin.  I have the most recent release of the BigTreeTech firmware here for the SKR V1.3 and I've scanned the config files and didn't see the errors they were referring to.  So, I'm afraid this might be a new problem.  (Which is why I started this new thread).

guru-florida, 2020-08-26:
It's possible the wiring itself overheated and one of the wires has failed. It's also possible the two different steppers have a different pinout. My Makerbot Replicator has a different pin-out than the new steppers I bought that were labelled as "MKS compatible". Apparently my makerbot are *not* standard. Also original wiring on the Rep2 was weak and low-amp. I use a continuity checker to verify the pinouts of the two bipolar coils and popped out the pins on the connector and rearranged.

The twitching shaft is a symptom of one coil not being powered. Easy way to check steppers is just to measure resistance of each coil, they should be equal. if one is lower, then you overheated the internal wiring of that coil and broke down the insulation thus shortening the electrical path. It's garbage. If coils still measure proper resistance then it's still ok...maybe degraded by overheating but would very likely still work. Now also measure the coil resistance (or continuity) from the TMC side of the wiring (with power off and without the TMC in socket). Jiggle the wires and connectors and make sure connection is maintained.

I doubt you burned out the TMCs since they have over-temp/current protection but if you were disconnecting steppers while current is on then you might have, or possibly moving the steppers/gantry while power is off but even this doesnt happen often. The 2209s can handle the 1.5A with ventilation but you shouldnt need anywhere close to that.

550mA or so should be plenty enough for extruders and is what I run. 800mA seems too high unless maybe at "volcano" fast extrusion rates. Keep in mind steppers heat exponentially in relation to mA, so find the floor and add some headroom.

Roop66, 2020-11-23:
Hi rrileypm. Did you manage to get this resolved. I have a similar problem.

rrileypm, 2020-11-24:
I had to upgrade to larger motors to handle the load. I did find out that I
had some cable issues. I swapped out the cables and the stuttering problem
stopped.

Roop66, 2020-11-24:
Thank you.
That's interesting. I think my motors only just lift the z axis gantry. If I apply a little upward pressure to assist it It works.  What upgraded motor did you chose? Did you have to modify current values in marlin for the steppe drivers?

Thank you 
Roop

shampine1, 2020-11-24:
I used a pair of constant force springs to balance part of the moving mass of the Z system on my big printer.
I also used a belt to tie the two Z motors together; this avoids having them get out of sync when the power is off.

Schlumberger-Private