bigtreetech/BIGTREETECH-SKR-V1.3 #414: SKR 1.4 Turbo Thermal Runaway Issue

Tougespirit33, 2020-09-05:
Hello,

After three or four solid prints my Ender 5 Plus with the mentioned main board halted mid print and went full tilt with bed and hot end heat All controls were unresponsive. Now when the machine is powered on it immediately heats the hot end and bed, as is evident by the illuminated blue lights on the bed mosfet. Attempting to reflash the firmware shows the appropriate FIRMWARE.CUR, however the problem persists. I also tried connecting the board to my pc via usb (I used the correct pin jumper) and pronterface. It works for about a minute then flags connection errors. I grabbed a new board and had the exact same issues. I have a suspicion that the firmware I used bricked both boards. Any help would be appreciated.

primerpop, 2020-10-05:
More or less same experience here on a Hypercube EVO, completed a first run calibration cube, and the hot end and bed now just run away from power-on.  Tried resetting EPROM to factory, re-uploaded last-known-good firmware.  No change.

Only some tweaks to config and all tests to that point were incrementally successful, geometries, BL-Touch, Sensor-less homing, fans, bed via mosfet and hot-end off the board.   Extrusion testing eventually yielded some decent results;  bed and nozzle responded as commanded.  Printed a 2cm^3 in PETG (240/80) cube that, given some stepping tweaks printed (not well, but this was the first print).   The print was completed and the machine never cut off the bed or hot end.   I pull the plug as the hot end pushes past 280 un-commanded.

My build based based on https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/tree/master/BTT%20SKR%20V1.4/Firmware/Marlin-2.0.x-SKR-V1.4-Turbo (Oct 2020 df17583)

zenturacp, 2021-01-11:
Just seen exactly the same issue :-/

Reported it as a bug on marlinfw but expect it's the board