bigtreetech/BIGTREETECH-SKR-V1.3 #415: Unable to control fan speed

DuckieCZE, 2020-09-06:
Hello. I have a printer with BigTreeTech 1.4 Turbo board running latest marlin firmware.
I want to control a fan that cools the print. The fan is connected to FAN0 (pin 2.03).
It stays on all the time. I cannot control it through Gcode.
I also have an extruder fan that I configured to turn on at 50degrees C.
What can be the problem? Is it even possible to control the speed?
Any help will be appreciated :-)
[marlinconfig.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/5179591/marlinconfig.zip)

I included configuration.h
                 configuration.adv.h
                 pins_BTT_SKR_common.h
[marlinconfig.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/5179591/marlinconfig.zip)

Arakon, 2020-09-06:
Only one of the fan ports is PWM controllable, the other is simply always on at full speed.

AnHardt, 2020-09-06:
> Is it even possible to control the speed?

No. It's a hardware limitation.

[image]

DuckieCZE, 2020-09-06:
> > Is it even possible to control the speed?
> 
> No. It's a hardware limitation.
> 
> [image]

Thank You for answering. I forgot to mention that the extruder fan is connected to HE1 (pin 2.4). So I should be able to control both of them, If the another one Is connected to FAN0 right? Both of these have PWM support.

DuckieCZE, 2020-09-08:
So I finnaly solved the problem. I defined the FAN_PIN as Pin 2.04. The pin 2.03 was burned out

DuckieCZE, 2020-09-08:
> Only one of the fan ports is PWM controllable, the other is simply always on at full speed.

Thank you for answering.

closed 2020-09-08 by DuckieCZE