bigtreetech/BIGTREETECH-SKR-V1.3 #417: SKR V1.4 Turbo

i3d-service, 2020-09-11:
Hello,

I have problems trying to compile the marlin firmware.

I want to compile the frimware Marlin-2.0.x-SKR-V1.4-Turbo but i have a lot of error and I don't know what to do anymore.

The error message:

ModuleNotFoundError: No module named 'platformio.managers.package':
  File "C:\Users\guimo\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 167:
    env.SConscript(item, exports="env")
  File "C:\Users\guimo\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\guimo\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 287:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\guimo\Documents\Marlin-2.0.x-SKR-V1.4-Turbo\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 12:
    from platformio.managers.package import PackageManager

Can someone help me?

olgerd5335, 2020-09-13:
hi !
I have the same problem.

timoleary, 2020-09-14:
Same issue with 1.3 firmware

tony1ant, 2020-09-16:
I'm having the same problems for 1.4 Turbo :((

cotestatnt, 2020-09-18:
Replace the file _buildroot\share\PlatformIO\scripts\common-dependencies.py_ taking from the original Marlin Github repo.
It was updated some days ago.

[https://github.com/MarlinFirmware/Marlin/blob/2.0.x/buildroot/share/PlatformIO/scripts/common-dependencies.py]

tony1ant, 2020-09-21:
ThkU
☆🌟☆🌟☆🌟☆🌟☆🌟
╔══╦╗──────╔╗
╚╗╔╣╚╦═╗╔═╦╣╠╗
─║║║║║╬╚╣║║║═╣
─╚╝╚╩╩══╩╩═╩╩╝
╔═╦╗  °☆°☆°☆°☆°☆°
╚╗║╠═╦╦╗           ☆°
╔╩╗║╬║║║        ☆°
╚══╩═╩═╝    ☆°
°🌟☆🌟☆🌟☆°