bigtreetech/BIGTREETECH-SKR-V1.3 #419: Blank screen when BLTouch connected to servo and probe pins on 1.4 Turbo board

Paulg5678, 2020-09-13:
I just installed a new 1.4Turbo board and all seems well except that when I connect all 5 pins of the BLTouch (3 servo pins and 2 probe pins), the LCD goes blank and the printer won't respond.  I have a creality BLTouch with red, blue and yellow wires for the servo and black and white wires for the probe.

Can someone please give me some guidance here?

thanks.

lunaemica, 2020-09-21:
I'm having the same issue with a BTT SKR E3 Turbo board.  Hoping we both get help on the issue?

MikeyD6, 2020-09-21:
i think it'll be the endstop wires. iwas using 3 wire endstops and this happened. i swapped em round (a few times) and it solved the problem. might be worth a try

Paulg5678, 2020-09-21:
Yeah, it turned out to be the servo wires actually.  I ended up having to switch the blue and red wire in the connector and then it worked fine.  This issue can be closed now I suppose.