bigtreetech/BIGTREETECH-SKR-V1.3 #423: [GUIDE] No update from SD solution?

borsegbr, 2020-09-25:
Possible solution to repair no update from SD
Contains guide and original dump from skr 1.4 turbo

[Flash skr1.4.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/5285159/Flash.skr1.4.zip)

gdanas87, 2020-10-20:
YOU ARE A LIFESAVER!!!! I am pulling my hair out for the past 4 days trying to figure out how I will flash new fw to my skr1.4, since I damaged the sd slot upon inserting the sd on initial startup when I unboxed the board (gotta admit, i have never had this happed to me before on any other board or device before in my life, so the sd slot is definately of totally crappy quality!!!)
Works just fine with an ftdi adapter btw, instead of the btt writer which i dont have. again, You are a lifesaver!!! Thank you so much, I have found a way to put my "broken" board to good use!!!

BerkhanB, 2020-11-02:
Just a quick note: I was going to follow this procedure but just tried another SD card before buying an FTDI adapter, it just worked. Try a new SD card before using this guide.

Frix-x, 2021-01-18:
After trying a lot of method to flash my new (and suspected dead from factory) SKR 1.4 Turbo without any luck, you're method finally make me able to do it !

Thank you very much !!!