bigtreetech/BIGTREETECH-SKR-V1.3 #424: SKR V1.4 Turbo died when 3.3V and 12V shorted

baymax2161, 2020-09-27:
Hello

I just got the SKR V1.4 Turbo board and was testing it on my 3D printer. I accidentally shorted 3.3V pin to 12V with a screw driver and the LPC1769 seems to have died. 
Could I just replace the LPC1769 and get it to work again? I have the soldering equipment and experience required to replace that chip. But I'm not sure if it has some sort of bootloader on it that I need to flash first.

I'd really appreciate any sort of help and suggestions. 
Thank you.

oldman4U, 2020-09-28:
Sure that not the fuses are burned?

radek8, 2020-12-31:
After replacing the CPU, you must load the bootloader

https://github.com/ardiehl/BTT_SKR_13_14_14T_SD-DFU-Bootloader
https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/issues/346
https://community.platformio.org/t/skr-1-4-bricked/12932/4