bigtreetech/BIGTREETECH-SKR-V1.3 #425: skr v1.4 turbo tmc2130 spi TMC CONNECTION ERROR

89oinotna, 2020-09-27:
Hi, i have a problem with this driver every time saying TMC CONNECTION ERROR

this is my configuration usin Marlin 2.0.6.1

[Marlin.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/5288282/Marlin.zip)

[image]

oldman4U, 2020-09-27:
Is there a red led between the 20 fuse and the power input?

89oinotna, 2020-09-28:
> Is there a red led between the 20 fuse and the power input?

@oldman4U yes there is

oldman4U, 2020-09-28:
And you have the tmc2130 defined in Marlin?

89oinotna, 2020-09-28:
@oldman4U 
yes, as can see it from the attached zip that contains configuration.h and configuration_adv.h

oldman4U, 2020-09-29:
And of course you set the jumpers right.

89oinotna, 2020-09-29:
yes they are right

tommyloon, 2020-10-17:
I have the same problem. Jumpers correct, Drivers correct, TMC2130 steppers. Software_SPI is on. 
TMC connection error on the screen. And errors on all drivers with the M122 command. 
Board en drivers are new and bought from the same shop as a SPI package.