bigtreetech/BIGTREETECH-SKR-V1.3 #426: No axys moving anymore

IIIxpHunterIII, 2020-09-28:
Hi there, I've just set up marlin so that I can home all axys and heat the printhead up. I was quite happy, since that was my first marlin build. After that I wanted to do a first testprint because I was curious what 32 bits can do qualitywise, therefore I PID'd the nozzle...aaand the my connection to pronterface was lost, the usb cable got warm and my usb hub was killed. After that I can still reach the Printer and control temps and fanspeed on Fan0, but all the axys are dead now. I'd assume it was my config mistake, but it shouldnt, since before entering M303 everything was fine. I hope anybody can help me with this problem.

Specs:
Driver 5 times FYSTEC TMC 2208 V1.2 in UART which by the way are not the problem, my trigorilla can still control em
SKR 1.4 Turbo
Marlin (used the official BIGTREETECH one)

oldman4U, 2020-10-01:
Not sure there is a setting in firmware which can cause this. I would expect hardware to be the reason for this.

IIIxpHunterIII, 2020-10-01:
Thanks for the help, i've contacted my reseller about it and hell take a look at my board.

oldman4U, 2020-10-01:
Perfect.

Please close the ticket once you do not need it anymore.

Thank you

closed 2020-10-07 by IIIxpHunterIII