bigtreetech/BIGTREETECH-SKR-V1.3 #428: Erratic Y Axis movement During Homing

CoilMonkey, 2020-10-04:
Ok ladies and Gents I have been googling without success.

Stock Ender 3 pro
BLTouch v3.1

I have upgraded(!) to an SKR 1.4, with TMC2209 stepper drivers. When I try to home all axis, I keep getting erratic movement on the Y axis as shown in the video attached ( I hope)

https://drive.google.com/file/d/1oDHLrTiW-7liidWqFVggUbnhkOoogdfZ/view?usp=sharing

Things I have tried
Swapping out stepper drivers - problem remains on Y axis only
Sensorless homing (diag pin removed)- problem remains on Y axis only
Mechanical endstop homing (Diag pin installed) - problem remains on Y axis only
inverting Y axis direction in firmware - problem remains on Y axis only
inverting logic on Y axis endstop - problem remains on Y axis only
y axis endstop pulldown enabled- problem remains on Y axis only
Y axis endstop pullup enabled - problem remains on Y axis only
Swapped X and Y stepper motors - problem remains on Y axis only
Swapped X and Y motor wiring looms - problem remains on Y axis only

are you all seeing the theme here?

I will post my config.h and config_adv.h but there really isnt a point as I think i have changed most settings out of pure desperation XD

config_adv
https://drive.google.com/file/d/10TP2jPfeoujnzmej8sENSuNul8bqqbGo/view?usp=sharing
config.h
https://drive.google.com/file/d/1bnZvDINGijhRl9nrnzzMgPXpDfKCSpS5/view?usp=sharing

Any ideas would be welcomed.

Thanks

oldman4U, 2020-10-05:
Have you tried to disable "fast homing" and to reverse y axis direction?

CoilMonkey, 2020-10-06:
I have tried reversing the Y axis direction, but haven't touched the fast homing.

I will try that later when I get home from work.

Thanks!

oldman4U, 2020-10-06:
Let me know

CoilMonkey, 2020-10-07:
Unfortunately that made no difference at all. Very strange.

oldman4U, 2020-10-07:
Does it happen also when you home only the y axis?

CoilMonkey, 2020-10-07:
Yes, sometimes when I use the move axis on Y it will move the wrong way too. Its very strange.

oldman4U, 2020-10-07:
Have you tried to swap the motor drivers of x and y to see if the problem moves?

CoilMonkey, 2020-10-08:
Yeah I tried that early on, I have a replacement board now on the way from btt.

oldman4U, 2020-10-08:
Hope this helps 👍🏻

Badhour, 2020-10-13:
Hi CoilMonkey,

I'm encountering a more or less similar problem but with the Z axis only if Y is homed before (strange). Tried more than 7 Marlin (2 pre-compiled and 5 different versions of Marlin), inverted the endstops to check, checked everything actually and I have the same conclusion as you : the problem must come from the board itself.
Anyway, how did you contact BTT for them to send you a remplacement board? I just spent every evening on it for more than 4 weeks by now...
Thank you very much

oldman4U, 2020-10-13:
@Badhour 

You have to contact the dealer where you bought the faulty hardware.

CoilMonkey, 2020-10-13:
I had to contact the original seller, it took a lot of back and forward and I'm still waiting on tracking I fo on the replacement parts.

I bought mine fro. The biqu ebay store so have had to go through the ebay conta t seller method.

I suggest you go back to your seller and Co tact them directly and explain your problem. Do not let them fob you off saying its a problem with your firmware settings.

Badhour, 2020-10-13:
Ok, thank you. Actually it was ordered on AE from the BTT or the BIQUStore (can't remember). I'll send them a message.
Thank you very much!