bigtreetech/BIGTREETECH-SKR-V1.3 #432: Two endstops on SKR 1.4 Turbo not working

IIIxpHunterIII, 2020-10-10:
I've got the problem, that the sanity check tells me:

#elif !HAS_Z2_MIN && !HAS_Z2_MAX
    #error "Z2_USE_ENDSTOP has been assigned to a nonexistent endstop!"    <--- Thats the actual error

And I really do not know what exactly he is complaining about.
I've tried to declare the Z_MAX in the .pins file, since that's been suggested in similar issues. that did let me compile it, but then the printer ripped the Z1 Screw out of the motorcoupling, Eg no Z1 endstop I suppose.

My config:
[Config.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/5358890/Config.zip)

IIIxpHunterIII, 2020-10-10:
#define Z_MAX P1_26 was missing in pins

closed 2020-10-10 by IIIxpHunterIII

Frittenkiller, 2020-10-22:
Can you just show me how to do it with the pens?

tca72, 2021-03-19:
Hello

This is normal as the SKR V14 or V1.4 turbo have only 3 endstops, 3 others connectors can be used as endstop but they have to be redefined in pins_BTT_SKR_V1_4.h :

you can define as : 
//
// Limit Switches
//

#define X_MIN_PIN                      P1_29  // X-STOP
#define X_MAX_PIN                      P1_26  // E0DET
#define Y_MIN_PIN                      P1_28  // Y-STOP
#define Y_MAX_PIN                      P1_25  // E1DET
#define Z_MIN_PIN                      P1_27  // Z-STOP
#define Z_MAX_PIN                      P1_00  // PWRDET

or use a more smart configuration as: 

#ifdef X_STALL_SENSITIVITY
  #define X_STOP_PIN                  X_DIAG_PIN
  #if X_HOME_DIR < 0
    #define X_MAX_PIN                      P1_26  // E0DET
  #else
    #define X_MIN_PIN                      P1_26  // E0DET
  #endif
// tca add {
#elif ENABLED(X_DUAL_ENDSTOPS) 
  #ifndef X_MIN_PIN
    #define X_MIN_PIN                      P1_29  // X-STOP
  #endif
  #ifndef X_MAX_PIN
    #define X_MAX_PIN                      P1_26  // E0DET
  #endif
// tca add }
#else
  #define X_STOP_PIN                       P1_29  // X-STOP
#endif

#ifdef Y_STALL_SENSITIVITY
  #define Y_STOP_PIN                  Y_DIAG_PIN
  #if Y_HOME_DIR < 0
    #define Y_MAX_PIN                      P1_25  // E1DET
  #else
    #define Y_MIN_PIN                      P1_25  // E1DET
  #endif
  // tca add {
  #elif ENABLED(Y_DUAL_ENDSTOPS)
  #ifndef Y_MIN_PIN
    #define Y_MIN_PIN                      P1_28  // Y-STOP
  #endif
  #ifndef Y_MAX_PIN
    #define Y_MAX_PIN                      P1_25  // E1DET
  #endif
  // tca add }
#else
  #define Y_STOP_PIN                       P1_28  // Y-STOP
#endif

#ifdef Z_STALL_SENSITIVITY
  #define Z_STOP_PIN                  Z_DIAG_PIN
  #if Z_HOME_DIR < 0
    #define Z_MAX_PIN                      P1_00  // PWRDET
  #else
    #define Z_MIN_PIN                      P1_00  // PWRDET
  #endif
  // tca add {
  #elif ENABLED(Z_MULTI_ENDSTOPS)
  #ifndef Z_MIN_PIN
    #define Z_MIN_PIN                      P1_27  // Z-STOP
  #endif
  #ifndef Z_MAX_PIN
    #define Z_MAX_PIN                      P1_00  // PWRDET
  #endif
   // tca add }
#else
  #ifndef Z_STOP_PIN
    #define Z_STOP_PIN                     P1_27  // Z-STOP
  #endif
#endif