bigtreetech/BIGTREETECH-SKR-V1.3 #435: USB Issues

rocketkittens, 2020-10-14:
I'm on my second board now, and it destroyed the USB controller on my PC.
The first board made the USB wire hot and burned the USB connector on the board.

This board is starting to feel unsafe, i have never had any USB issues with my computer or any other peripheral devices before this incident.

I learned that the first board i had on USB jumper mode. And the second one i had correctly set to VDD jumper mode.

But i am still distressed about why this damage is occurring, does it have anything to do with limit switches perhaps?

Please advise

rocketkittens, 2020-10-14:
Just as a warning to other users, i would highly avoid using BTT products and the SKR1.4, it has cost me a lot of time and money. Bought 2 boards already with the result being spending more money on a new computer because it destroyed my motherboards USB controller.

I don't think anyone will be addressing this issue. There is a potential fire hazard when plugging in USB, this board has quality control issues.

Arganaz, 2020-10-29:
This might help with that problem.
https://community.octoprint.org/t/put-tape-on-the-5v-pin-why-and-how/13574

rocketkittens, 2020-10-31:
Tape is not a helpful solution, i am trying to understand why the board killed my computer and how to diagnose and prevent it.

I see an issue that may be the same as what i experienced:
https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/issues/443

rocketkittens, 2020-10-31:
I have noticed that when i plug in USB into the same mobo the PSU fan instantly runs louder, so it appears the issue still exists.

The power mode pin is set to VDD, i was expecting that it would be possible to connect the printer to a computer via USB while powered by 12v but it appears that this is not possible.