bigtreetech/BIGTREETECH-SKR-V1.3 #440: TMC Connection Error on SKR V1.4 Turbo with TMC2208 drivers

mcdermottdylan16, 2020-10-25:
I have been trying to get my Ender 3 v1 running with the BTT SKR v1.4 Turbo board and BTT TMC2208 v3.0 drivers in UART mode. No matter what I try I keep getting TMC CONNECTION ERROR. I have also connected over USB to my computer and enabled the TMC_DEBUG to get some extra information through Pronterface, but when running a M122 I get all false/error on each of the four axes. 

Pronterface M122 output:
[https://i.imgur.com/xSCFamE.png](url)

I followed the directions explicitly when making firmware changes required for both the mainboard and drivers. Thinking it's possible I did something wrong I have referenced several tutorials and I can still not figure out what the problem is as it appears I'm doing everything correctly. I have tried using the Marlin-2.0.x branch for SKR V1.4 turbo that BTT provides on this github as well as the main bugfix branch off the Marlin github i.e. bugfix-2.0.x to no avail. 

I have set up the jumper caps as specified in the SKR v1.4 turbo documentation for the UART configuration, and the three configuration solder pads should be of no issue for the BTT TMC2208 v3.0 as they are supposed to be shipped with the UART enabled configuration. I've inspected each driver and they appear to have a 0 Ohm resistor shorting the appropriate two config pads for UART as specified in the documentation, which is what I expected. 

I am pretty new to working with 3D printers so maybe this is normal or to be expected, but the motors still work. I can send commands on Pronterface and move them to a specified position, autohome works, I can control the axes from the LCD, but I still have the TMC CONNECTION ERROR. I can go into the advanced settings menu on the LCD, click TMC Drivers, and see that every driver is enabled with Stealth Chop and each drivers current is set. 

At this point I'm at a loss as to what is my issue. Maybe I am doing something wrong in the firmware, or maybe my mainboard is damaged. Maybe there is actually nothing wrong but for some reason I'm getting thrown an error? I don't know, so I will link my config.h below and hope I can get some help here. Thanks!

My config.h:
[Configuration.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/5433954/Configuration.zip)

tribayf, 2020-11-04:
I have the same issue. Haven't figured it out yet.

mkotyk, 2020-11-12:
I was seeing a similar issue with my BTT TMC2209 and SKR 1.4. .. I eventually tracked the problem down to the 20K  UART pulldown resistor (R5 on the driver boards).  My solution was to remove that resistor, and then I was able to get bi-directional communication working.

https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/issues/462

tribayf, 2020-11-14:
@mkotyk @mcdermottdylan16 In my case it was a faulty SKR 1.4 (turbo). I was only getting the connection error for the X driver, but swapping the drivers around had no effect. I ordered a new SKR 1.4 and the issue was resolved.

i3creations, 2020-12-02:
I too am getting this error.  I was testing using USB 5v power.  When I tied in my 24V PSU the error went away.

mcdermottdylan16, 2020-12-04:
I could never resolve my problem with the TMC Error. I tried a completely new SKR v1.4 Turbo board with original TMC2208 drivers and the problem persisted. I then tried to replace them with TMC2209 drivers. The only changes I made to the firmware was to change my drivers from TMC2208 to TMC2209 and then I had a working UART mode on my drivers. I don't really know if this means all my drivers were dead or if this is a blunder on Big Tree Techs part for making bad design choices. My problem was not that I was using USB power, although I know that is a common issue that causes this error.

closed 2020-12-04 by mcdermottdylan16

Asphyriax, 2021-01-15:
I have a skr 1.4 turbo, When i had the same tmc error, there were only 4 STEPPENMOTORDRIVERS CONNECTED.

when i snipped one of the side leads to put it in the same mode(check internet seems to be a common thing if you dont want to use sensorles homing) and plugged it in the error went away.

Means that pluggin in all steppen motor drivers is an solution.

Rage997, 2021-05-01:
Same exact problems with SKR V1.4 turbo and TMC2208 drivers - however, the printer works flawlessly so I don't know what to think