bigtreetech/BIGTREETECH-SKR-V1.3 #443: skr 1.4 turbo is very weak

labotecno, 2020-10-26:
i have 2 board with the same problem , i try to update marlin firmware from usb and get died and burn the sd memory too .
bilan i cant use it because with 12v 24 v jumper is not reconize anymore the serial usb , only with 5 v jumper is working but dont have enough power to make it working the printer ... i look for the board and i feel heat on the part of pcb looking is a short circuit ...
i have skill to solder 
i need the shematic of this board 
if someone have problem like me can you add your issue 
bbt can you give me a hand to repair the board ? 
thank you

rocketkittens, 2020-10-31:
I would like BTT to explain if it is safe to power the board with 12v and connect the board to USB at the same time because thats where it ends with smoke for me.

labotecno, 2020-10-31:
> 
> 
> I would like BTT to explain if it is safe to power the board with 12v and connect the board to USB at the same time because thats where it ends with smoke for me.

usualy its working , and reconize the port com to the computer , be carefull to never use usb mode jumper !i think the problem is about this line  ... i try to see what is the smd component is frying .... to repair two board with the same issue .

Rincewind08, 2021-03-29:
for me it was the proximity of the plug and the little ic above it for communications, somehow the plug overheated so bad  (under serious air cooling  ) that the 6 leg ic above the turbo's usb port's legs melted.