bigtreetech/BIGTREETECH-SKR-V1.3 #447: SKR 1.4 Turbo, TFT35 v3.0 in Touchscreen Mode and M997 disables SKR USB on reset

DodgeDeBoulet, 2020-10-31:
I'm having an issue with a BTT SKR 1.4 Turbo and TFT35 v3.0. Issuing an M997 "reset" command when the display is in Graphic/TFT mode will reset the board, but it does not re-establish USB connectivity to the host; /dev/ttyACM0 does not become available again. In "Marlin" mode the reset works normally.

I've running Marlin 2.0.7.2, and have tried 2.0.7.1 and 2.0.6.1 in an attempt to determine if Marlin is the culprit, but the issue happens on all 3 versions.

Note that if there is a firmware.bin file on the SD card, the M997 does result in it being flashed. But after the flash completes, USB connectivity to the host (in my case a Raspberry Pi 4) is not re-established.

A power cycle of the board returns operation to normal and /dev/ttyACM0 becomes available again.

oldman4U, 2020-11-04:
Seems you have a twin brother: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/893

DodgeDeBoulet, 2020-11-05:
Closing this issue as it is being tracked more completely elsewhere (and boy am I glad I'm not alone!)

closed 2020-11-05 by DodgeDeBoulet

oldman4U, 2020-11-05:
I will try to take care asap.

Thank you