bigtreetech/BIGTREETECH-SKR-V1.3 #450: X axis stalls wont travel / home

iHover01, 2020-11-01:
I just installed SKR V1.4 Turbo board with TMC 5160 stepper drivers on all axis.  when I home all the axis work fine except for the x axis. it begins moving to home but stops traveling after moving a 10-20 mm. If I move it close to the switch it it homes but it I try bed leveling following that it has the same behaviors, it moves a few mm and then stops. I have all the TMC settings for this axis setup the same as the Y axis which works without any issues.

mpapam, 2020-11-14:
Have been trying for hours and hours to make my TMC5160 to work on SKR without any success, no movement at all. Can you please share your configuration.h and configuration_adv.h files along with your hardware setup?
Regarding your issue I suggest you exchange drivers with Y or Z axis to see whether it is a faulty driver issue

iHover01, 2020-11-15:
Sorry though I closed this. It appears to have been an issue with a cable

closed 2020-11-15 by iHover01