bigtreetech/BIGTREETECH-SKR-V1.3 #451: X axis stalling  only moving short distance

iHover01, 2020-11-01:
I just installed SKR V1.4 Turbo board with TMC 5160 stepper drivers on all axis.  when I home all the axis work fine except for the x axis. it begins moving to home but stops traveling after moving a 10-20 mm. If I move it close to the switch it it homes but it I try bed leveling following that it has the same behaviors, it moves a few mm and then stops. I have all the TMC settings for this axis setup the same as the Y axis which works without any issues.
config files
config adv.
https://drive.google.com/file/d/1RlC1totTHm6aJ1mVnMK3pcCO04_5FKyX/view?usp=sharing
config
https://drive.google.com/file/d/1fIDkXxr-6AUhYGCYH5F_0rOp1J1Fol8h/view?usp=sharing

jasonfox7, 2020-11-02:
Have you checked the current going to your stepper? I just had a similar issue and it was undercurrent from the stepper drivers.

iHover01, 2020-11-02:
The current is set the same as the y axis and that seems to be working just fine.  I think I found the issue, it didnt like the extension cable I had on the x axis. Not sure why , Ive had that extension on since I originaly setup the printer and its never had an issue with the cr10 board.

eric123v, 2020-11-04:
After literally a week of working on this every night, I found this and changed it in my config_adv.h file. HOMING WORKS! I was having the same problem as you.  

I uncommented #define ENDSTOPS_ALWAYS_ON_DEFAULT  around line 500

Now on to all my other problems - I hope this works for you.

// @section homing

// If you want endstops to stay on (by default) even when not homing
// enable this option. Override at any time with M120, M121.
#define ENDSTOPS_ALWAYS_ON_DEFAULT

iHover01, 2020-11-04:
My issue turned out to be a problem with one of the cables. I had an extension cable on that stepper motor once I removed it everything worked fine.

iHover01, 2020-11-04:
Turned out to be a problem with the cabling to the stepper motor. I was using a cr10 extension cable set and for some reason the new boards drivers didnt like it.  Once I removed it everything worked

closed 2020-11-04 by iHover01