bigtreetech/BIGTREETECH-SKR-V1.3 #454: Hotend not "hotting"

eric123v, 2020-11-03:
I must have updated this firmware 25 times - I am confident that I have spent over 40 hours just tinkering and going through different settings on the configuration files - nothing I do gets this to work.  
* I can't get the hotend to heat (I also can't get my homers to home, but one step at a time!).*

Can anyone look at this config file and tell me if I have buggered something up in it?  I've tried on my own, failed, rinsed, and repeated...ad nauseum - I am at my wits end and if I had hair, I'd be pulling most of it out. 
Here is my setup: 
- Ender 5 Base model (not plus)
- Marlin 2.0 Firmware 
- BTT SKR V1.3 set up in UART Mode with 2208 Drivers
- TFT 24 v1.1 using older firmware
- Firmware using Ender 5 Config.H and adv.h examples from BTT (modified stepper type (TMC2208) and direction for X/Y, display (...Full discount...) 
- Upgraded to Meanwell PSU

All endstops and hotend was working with stock board and display. I was really trying to use the stealthchop drivers to quiet the printer down so I could use it in my office while WFH.
I have a Creality brand BL Touch, but it is not yet configured or plugged in until I get these other issues working.  

[Configuration.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/5479305/Configuration.txt)
[Configuration_adv.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/5479306/Configuration_adv.txt)

If anyone can help - specifically with the hotend, but also with the endstops, I'd be incredibly grateful!!  Thanks - E

oldman4U, 2020-11-04:
Have you tried to measure the current which is provided at the hotend while heating up?

eric123v, 2020-11-06:
Hey @oldman4U ...thank you for getting back to me. My problem was a really stupid one on my part. Wiring issue into board. Got it working and now working through trying to adjust the steps.

oldman4U, 2020-11-06:
👍🏻

Please help the community and close the ticket once you do not need it anymore. Thank you and happy printing

closed 2020-11-06 by eric123v