bigtreetech/BIGTREETECH-SKR-V1.3 #456: SKR V1.4turbo doesnt connect thru USB after VSCode upload

Kscott73, 2020-11-05:
I moved the jumper to USB pins, then plugged USB cable into the computer. The board loaded up and connected to COM3.
I uploaded Marlin firmware thru VSCode
[firmware.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/5491391/firmware.zip)
now it wont connect but powers up the board

oldman4U, 2020-11-05:
Which serial scare defined in configuration?

Kscott73, 2020-11-05:
i pulled the sd card and loaded file straight to the card. inserted card back in the skr board and plugged it in to the usb port. and guess what, it found the COM3 port.

side: wont connect to pronterface
[ERROR] Could not connect to COM3 at baudrate 250000:
Serial error: could not open port 'COM3': WindowsError(5, 'Access is denied.')

oldman4U, 2020-11-05:
Please help the community and close the ticket once you do not need it anymore. Thank you

Kscott73, 2020-11-05:
sry

closed 2020-11-05 by Kscott73

oldman4U, 2020-11-05:
No problem. Thank you and happy printing