bigtreetech/BIGTREETECH-SKR-V1.3 #457: installed skr1.4 turbo in siderwinder x1

Kscott73, 2020-11-06:
hey 
i installed a BTT SKR 1.4 turbo into Sidewinder X1 today,  I found that the dual z and the extruder wires where crossed. And I had the extruder inverted so it was unloading when I selected load. but they were easy fixes. My problem is when I pressed "Home",  X,Y,Z went to the farthest positions.

oldman4U, 2020-11-07:
Hi

You can change the homing direction in Marlin.

Kscott73, 2020-11-07:
I need more. I am trying to run stallguard on x,y,z and it causes alot of errors

oldman4U, 2020-11-07:
First problem solved?

Which error?

Kscott73, 2020-11-07:
i started over, using marlin's bugfixed version.  i think problem was because i changed #define INVERT_X_DIR, i forgot to invert #define X_HOME_DIR

my coordinates are still inverted. extruder head moves in the right direction but home is 300,310,400
[image]

Kscott73, 2020-11-10:
i finally managed to get motors and coodinates correct
today i had help to get screen updated and correct(thank you oldman4)
so the only thing left is sensorless homing

[image]

oldman4U, 2020-11-10:
Good.

But Marlin is not recognized and TFT firmware is completely outdated.

Kscott73, 2020-11-13:
well;
marlin finally did get recognized
and finallly gave up on sensorless home, cut the pins and put the endstops back on
👎 :
but i have working machine. i will come back to sensorless home 6-12 months when it is more reliable.. just need 4 steppers

closed 2020-11-13 by Kscott73