bigtreetech/BIGTREETECH-SKR-V1.3 #458: SKR V1.4 Turbo doesn't connect via USB

Trytas, 2020-11-08:
Hey guys,

I got myself a SKR 1.4 turbo and I'm allready 3 days working on the damn thing and can't get the USB Connection to work
In my opinion the firmware is setup the right way and should work

https://www.dropbox.com/s/wqu2cb1aiu05r8g/firmware.bin?dl=0

I tried everything now I could find on YT

Trytas, 2020-11-08:
https://www.dropbox.com/sh/9aahv5hq3sb2f3z/AACnUjZ77FrZzDNp9NcebgmOa?dl=0

Here is the complete Folder if anyone needs it

tarifari, 2020-11-26:
maybe your usb "thing" is burned. I have two boards with same issues.

https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/issues/50

Maybe this helps. Try to solder out D3.

I´ll try it today and will report my results

Trytas, 2020-11-26:
Would be great
Would like to use the USB because I'm not really in the mood the upload everything on the SD and put it into the printer

tarifari, 2020-11-26:
is it just the usb connection? can you control the printer? my display also says "no printer attached" so i dont know if its just the usb connection. Cant control the printer too

Trytas, 2020-11-27:
The printer work's just fine if controlled over the Display.
So I need to put the gcodes on my SD card and then start the print over the Display.
At the moment I'm using the stock CR-10 display