bigtreetech/BIGTREETECH-SKR-V1.3 #459: [SKR 1.4 Turbo][Marlin 2.0] Magic smoke release

MS-Dzo, 2020-11-08:
Hi. i used my SKR 1.4 Turbo for the first time today. I've updated the firmware to Marlin 2.0 and managed to get some nice quiet prints. while manually doing a filament swap, the board started beeping rapidly and i saw some smoke release. i shut the power off as fast as possible and inspected the board up close. a single smd component burned near Y1 (i assume a capacitor ?)

pic of the fried component :
[image]

I've also found some schematics online for the SKR 1.4, and i assume the turbo is mostly the same ?
[image]
if they are the same, then the burned component must be one of those (minus Y1,C16 and C17 since they're part of the oscillator and should be as close together as possible, if i remember my electronics classes correctly ?)
[image]

So the questions are : 
-What happened?
-how can i prevent it in the future?
-how can i repair this ?
- I previously had an SKR 1.3 who failed in exactly the same circumstances, at EXACTLY the same equivalant spot
[image]
(pic taken from a reddit thread about another person who experienced something similar to me. the comments mention that we are not alone)

it seems to me that all those component were connected to (at most) 3.3V, so how the hell did a higher voltage reach that part of the circuit ??

I bought another board on Amazon, hoping that if it blows up on its own i'll get a refund.

MS-Dzo, 2020-11-16:
Update, i hooked up the new board and it instantly fried at exactly the same place. I'm starting to believe i'm not hooking up the board correctly, but i have no clue what i'm doing wrong. i'm using the regular 12v supply from an old CR10 that i've yet to check on an oscilloscope (maybe there's some current leakage somewhere ?? or some voltage spikes when starting up ??)
Other than that, since i'm using a CR-10, the wiring is pretty straight forward. I'm starting to believe i'm missing something obvious, but if that was the case why would the first two boards die after weeks of use ??

This whole thing has left me very confused.

noonetry, 2020-11-23:
Hi.
 Looking at [silkscreen image](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/BTT%20SKR%20V1.4/Hardware/BTT%20SKR%20V1.4SY.pdf) I see FB3 marking where your part is burned. From [schematics](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/BTT%20SKR%20V1.4/Hardware/BTT%20SKR%20V1.4-SCH.pdf) looks like this is ferrite bead, some small inductance that decouples GND (signal and power ground) and VSSA (analog ground for ADC). It usually have impedance parameter measured at tens or hundreds MHz, for example 120 Ohm @ 100 MHz.
 My logic says that FB3 can be burned due to abnormal voltage difference between GND and VSSA.
 I have searched VSSA text in schematics and found it only near thermistors connectors, especially those pins, which connected directly to "-" pins. Also there are strange RV9/10/11 parts, by product number and by marking "RV" they are varistors, but symbols are similar to TVS diodes.

 Check your thermistors "-" wires.

[image]

MS-Dzo, 2020-11-25:
I am really thankful for your answer. I will redo and replace the wiring because the wire management at the time is... minimal and i soldered extra length of wire to the existing default CR10 cables . I will also check if there's a short from the thermistor to the hotend, and from the thermistor to the heated bed but i haven't touched these places much so idk why something would have happened suddenly.
I also checked my power supply on an oscilloscope with a load of 500 ohm at startup and i saw an overshoot of ~1.5V but no "undershoot" (i thought maybe an oscillation was happening at startup, pulling current through FB3 from the mcu/VSSA ?? idk). 

With your explanation there is only two possible things happening : either my ground is shorting to something, or the thermistor is shorting with something. either way, i will be searching thoroughly.

Thank again for answering.

javata70, 2021-09-24:
@MS-Dzo Did you find the root cause of your problem?
I burned 2 SKR board when I've connected the Thermistor

[image]

arhodiewithsoul, 2022-05-10:
I thought I was the only one who had encountered this fault:

I had a BTT SKR v1.4 control board installed in a Voron CoreXY printer until early February. It was working perfectly fine with the Klipper firmware installed and many printing hours (200+ hrs in service). I had just initiated a new print which I then cancelled and started re-homing the printer – but during the homing procedure, the x and y motors stalled, the PL-05 sensor flashed and the hot end fan stopped. It seemed like the power on the machine flashed. The motors then started making a noise like they were trying to move against each other, so I immediately cut the power. On inspecting the board, I could see that FB3 had blown.

[image]

I installed an old Ramps setup and the motors, drivers etc worked fine and completed some test prints without fault, so I opened a new from box SKR v1.4 and installed that. That board was working up until last week (about 300 hrs in service) when following a filament change, I was cleaning the nozzle. The heater cartridge was not powered, but the printer was still powered up and idle. Unfortunately, I had the same sudden power flicker and the printer became non-responsive. On inspecting the board, I could see that FB3 had again blown.

[image]

I cannot find any exposed wiring on the hotend heater or thermistor, but somehow it appears that FB3 is failing due to shorting or overvoltage. 

I'm at a complete loss as to what is causing it.

I'm also down to my last v1.4 board, so if this one suddenly blows, I'm just going back to Ramps.

MS-Dzo, 2022-05-10:
Hello, I believe the initial theory that some kind of current going from VSSA to GND is valid. Saddly in college so i don't have access to my 3D printer / workbench for a while so i can't test it, but since that inductor can only be used to filter digital noise, i think it's safe to assume that we can short it. 

If you have access to a reflow oven/ you have steady hands, you can probably short it with no real issues. It would be better to replace it with a bigger impedance. 
something like the BLM/BLA/BLT should be good.
https://www.murata.com/~/media/webrenewal/support/library/catalog/products/emc/emifil/c31e.ashx.

By the way, the board still ended up burning even with the Hemera. :v

arhodiewithsoul, 2022-05-16:
Thanks for your reply and sorry to hear that it still failed with the Hemera. 

Unfortunately, I don't have access to a reflow oven and my technical abilities in this regard are somewhat limited. What I did want to do was add my experience to your thread in the hope that others with more complete knowledge (possibly BTT technical support) would be able to respond to definitively identify the factors which caused the fault... and extrapolate from there to offer a solution. 

Both of my boards presented FB3 blown, the first seems to have destroyed FB2 along with it. In theory, I can replace them with the same 120R SMDs they were originally fitted with and then see what damage (if any) the chip has sustained. I may, with luck, be able to revive the second board - the first one *looks* terminal.

I still do not understand what caused it though. In simple circuits, I would expect to see an obvious bridge between higher/lower voltage and resulting short. In this circuit (??) 

I have an ABS mounted hotend with the Heater Cartridge and Thermistor as the only two electrically active components. They are mounted on opposite sides of the aluminium Heat Block. The Cartridge has no exposed wires that could come into direct contact with the Heat Block, so I am initially discounting that as a conductor between the components. The Thermistor and Heater Cartridge are insulated from each other along the full length of their respective looms and terminated in male plugs into their respective sockets. 

What happens on the board circuit and *if* there is a drain/ghost in the shell which would allow a circuit to complete or earth through the Thermistor GND - I'm not qualified to diagnose. 

The obvious short would present itself at first operation - power the printer on/ activate the hotend and puff of magic smoke. That would indicate faulty wiring. Operating the printer for many hours before a fault presents would suggest a faulty component. 

Please feel free to question/correct any of the above ;-)

arhodiewithsoul, 2022-05-18:
Reply from BTT Technical Support:

Hi ，
Thanks for your detialed information.
From the picture ,  it is result from the leaking electrict by the thermistor cause the FB3 burning. 
I check your order number at our platform it inditcat that  the skr v1.4 has out of warranty.
Thanks for your understanding.

...

Hooikraal3D, 2022-12-06:
Ok, the heater cartridge has an internal element contact to the case, it's possible I over tightened the securing screw while clamping it, in defense I did only 'nip' it up. A contact to the heater block from the 'hot' side of the thermistor puts 12volts down the 3.3volt side which is carried to ground via LV5, a tiny surface mount inductor. Once that blows the 12volts finds its way into the CPU which fries it.