bigtreetech/BIGTREETECH-SKR-V1.3 #461: SKR V1.3 + FYSETC TMC2209 V2.1 steppers not moving

NSchnitzler, 2020-11-10:
Hello everybody,

i have a non-pro SKR V1.3 and FYSETC TMC2209 V2.1 stepper drivers, the board is flashed with the marlin fw. 
All jumpers below the stepper drivers are removed, the UART Jumpers are set. The communications seems to work, here is the TMC Debug output:
```
SENDING:M122
axis:pwm_scale/curr_scale/mech_load|flags|warncount
		X	Y	Z
Address		0	0	0
Enabled		false	false	false
Set current	1200	1200	1200
RMS current	1160	1160	1160
MAX current	1636	1636	1636
Run current	20/31	20/31	20/31
Hold current	10/31	10/31	10/31
CS actual	10/31	10/31	10/31
PWM scale
vsense		0=.325	0=.325	0=.325
stealthChop	false	false	false
msteps		16	16	16
interp		true	true	true
tstep		max	max	max
PWM thresh.
[mm/s]
OT prewarn	false	false	false
triggered
 OTP		false	false	false
pwm scale sum	12	12	12
pwm scale auto	0	0	0
pwm offset auto	36	36	36
pwm grad auto	14	14	14
off time	4	4	4
blank time	24	24	24
hysteresis
 -end		2	2	2
 -start		1	1	1
Stallguard thrs	0	0	0
uStep count	40	40	40
DRVSTATUS	X	Y	Z
sg_result	0	0	0
stst
olb
ola
s2gb
s2ga
otpw
ot
157C
150C
143C
120C
s2vsa
s2vsb
Driver registers:
		X	0x80:0A:00:00
		Y	0x80:0A:00:00
		Z	0x80:0A:00:00
Testing X connection... OK
Testing Y connection... OK
Testing Z connection... OK
```
I have tried stealthChop and spreadCycle mode, but the steppers (lying on the table, without any mechanical connection) are just vibrating, not turning. I have checked the Wiring so that one coil is connected to the outer pins and the other to the inner pins of the motor connector of the board.  After hours of investigation, i ran out of any idea how to make the setup work.
Does anyone have an idea how to solve this issue?

thanks in advance!

NSchnitzler, 2020-12-17:
I have found out, that the z-axis works as expected. But if i switch the driver, stepper and cable to x or y, the stepper doesn't move.

Has anyone an idea how to solve this?!

radek8, 2021-01-10:
I would try to swap 2 pins on the motor connector
for example, Creality engines have interchangeable terminals with NEMA engines

NSchnitzler, 2021-01-10:
i have tried all combinations of the motor wiring.