bigtreetech/BIGTREETECH-SKR-V1.3 #470: SKR 1.4 Turbo will not compile

f-wilhelm, 2020-11-25:
I want to update my Anycubic Kossel Plus with the SKR 1.4 Turbo, have done all settings like described in the manual, but it doesn't work. Can anyone give me a hint or help me, please?

Processing LPC1769 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.3.zip; board: nxp_lpc1769; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
In file included from d:\marlin-2.0.x\marlin\src\inc\MarlinConfigPre.h:37,
                 from d:\marlin-2.0.x\marlin\src\inc\marlinconfig.h:28,
                 from buildroot/share/PlatformIO/scripts/common-dependencies.h:31:        
d:\marlin-2.0.x\marlin\src\core\boards.h:395:64: error: operator '==' has no right operand
  395 | #define _MB_1(B)  (defined(BOARD_##B) && MOTHERBOARD==BOARD_##B)
      |                                                                ^
d:\marlin-2.0.x\marlin\src\core\macros.h:159:29: note: in expansion of macro '_MB_1'
  159 | #define _DO_1(W,C,A)       (_##W##_1(A))
      |                             ^
d:\marlin-2.0.x\marlin\src\core\macros.h:171:28: note: in expansion of macro '_DO_1'
  171 | #define __DO_N(W,C,N,V...) _DO_##N(W,C,V)
      |                            ^~~~
d:\marlin-2.0.x\marlin\src\core\macros.h:172:28: note: in expansion of macro '__DO_N'
  172 | #define _DO_N(W,C,N,V...)  __DO_N(W,C,N,V)
      |                            ^~~~~~
d:\marlin-2.0.x\marlin\src\core\macros.h:173:28: note: in expansion of macro '_DO_N'
  173 | #define DO(W,C,V...)       _DO_N(W,C,NUM_ARGS(V),V)
      |                            ^~~~~
d:\marlin-2.0.x\marlin\src\core\boards.h:396:19: note: in expansion of macro 'DO'
  396 | #define MB(V...)  DO(MB,||,V)
      |                   ^~
d:\marlin-2.0.x\marlin\src\pins\pins.h:420:7: note: in expansion of macro 'MB'
  420 | #elif MB(BTT_SKR_V1_4_TURBO)
      |       ^~
In file included from d:\marlin-2.0.x\marlin\src\inc\marlinconfig.h:34,
                 from buildroot/share/PlatformIO/scripts/common-dependencies.h:31:
d:\marlin-2.0.x\marlin\src\pins\pins.h:747:6: error: #error "Unknown MOTHERBOARD value set in Configuration.h"
  747 |     #error "Unknown MOTHERBOARD value set in Configuration.h"
      |      ^~~~~
CalledProcessError: Command 'arm-none-eabi-g++.exe -D__MARLIN_FIRMWARE__ -DU8G_HAL_LINKS -D__MARLIN_DEPS__ -w -dM -E -x c++ buildroot/share/PlatformIO/scripts/common-dependencies.h' returned non-zero exit status 1.:
  File "C:\users\friedrich wilhelm\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 168:
    env.SConscript(item, exports="env")
  File "C:\Users\Friedrich Wilhelm\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Friedrich Wilhelm\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 287:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "D:\Marlin-2.0.x\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 286:
    apply_features_config()
  File "D:\Marlin-2.0.x\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 118:
    if not env.MarlinFeatureIsEnabled(feature):
  File "C:\Users\Friedrich Wilhelm\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "D:\Marlin-2.0.x\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 262:
    load_marlin_features()
  File "D:\Marlin-2.0.x\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 250:
    define_list = subprocess.check_output(cmd, shell=True).splitlines()
  File "C:\Users\Friedrich Wilhelm\.platformio\python3\lib\subprocess.py", line 411:
    **kwargs).stdout
  File "C:\Users\Friedrich Wilhelm\.platformio\python3\lib\subprocess.py", line 512:
    output=stdout, stderr=stderr)
================================================================================================== [FAILED] Took 1.85 seconds ================================================================================================== 

Environment    Status    Duration
-------------  --------  ------------
LPC1769        FAILED    00:00:01.853
============================================================================================= 1 failed, 0 succeeded in 00:00:01.853 =============================================================================================The terminal process "C:\Users\Friedrich Wilhelm\.platformio\penv\Scripts\pio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

oldman4U, 2020-11-25:
Hi delete everything inside the folder (WINDOIWS) C:\Users<USER>.platformio and try again.

f-wilhelm, 2020-11-26:
Hi, thank you, but now nothing happens:

Executing task: d:\Daten\3D\BTT\Marlin-2.0.x\pio.exe run <

The terminal process failed to launch: Path to shell executable "d:\Daten\3D\BTT\Marlin-2.0.x\pio.exe" does not exist.

Terminal will be reused by tasks, press any key to close it.

oldman4U, 2020-11-26:
You need PIO and VSC support. Repeat what I wrote or check Google.

f-wilhelm, 2020-11-26:
Hi, I just had forgotten to reload VS, now it works, thank you.

closed 2020-11-26 by f-wilhelm