bigtreetech/BIGTREETECH-SKR-V1.3 #472: About the DCDC module : request for the SKR 1.5 or 2.0 ?

yet-another-average-joe, 2020-12-02:
I also got the DCDC bridge when I purchased a 1.4 Turbo. Because why not ? Even if I will probably not build a lightshow around the printer (but who knows). Reading the documentation : no more than 30 Neopixel. According to Adafruit, one Neopixel can eat up to 60mA. This means 1.8 Amps. The converter chip is a MP1584, that can give up to 3 Amps in ideal conditions (in terms of PCB surface, copper - 2 oz -  and thermal vias). We are far from these conditions, and these conservative recommandations seem reasonable.

Why couldn't BBT make a motherboard with a beefier buck (say 5A), that could also power a Raspberry Pi ? And why not a pinheader for the Raspberry Pi GPIO with GND, +5V, TxD and RxD, + some other I/O ?

An OctoPrint printer would be so easier and cleaner !

When will there be an OctoPi friendly printer motherboard ??? Tired with all that sh*tty wiring !

Hipsta69, 2020-12-02:
Check. Out the BTT GTR, that has a 40 pin header to plug a RPi straight in 😁

yet-another-average-joe, 2020-12-02:
Oh yes ! I didn't pay attention to the RasPi connector because I understood it was designed for tool changers. Didn't even look at the silkscreen (Raspberry Pi unpopulated pinheader...)

The communication on the Aliexpress store is really bad : 11 axis, multi-color ! I didn't read further. Raspberry Pi is not even in the "specification". Have to scroll down a lot before you can read "Support Raspberry Pi".

And something between SKR 1.4 and GTR ? For random people playing with random printers.
A basic breakout header for RasPi would require only 2x5 pins and 3 Amps...

james-fry, 2021-03-29:
> I also got the DCDC bridge when I purchased a 1.4 Turbo. Because why not ? (and also for the BLTouch...). Even if I will probably not build a lightshow around the printer (but who knows). Reading the documentation : no more than 30 Neopixel. According to Adafruit, one Neopixel can eat up to 60mA. This means 1.8 Amps. The converter chip is a MP1584, that can give up to 3 Amps in ideal conditions (in terms of PCB surface, copper - 2 oz - and thermal vias). We are far from these conditions, and these conservative recommandations seem reasonable.
> 
> Why couldn't BBT make a motherboard with a beefier buck (say 5A), that could also power a Raspberry Pi ? And why not a pinheader for the Raspberry Pi GPIO with GND, +5V, TxD and RxD, + some other I/O ?
> 
> An OctoPrint printer would be so easier and cleaner !
> 
> When will there be an OctoPi friendly printer motherboard ??? Tired with all that sh*tty wiring !

Im running 48 x WS2812b LEDs from my 5v dcdc with no problems.
I just dont run them at full brightness.