bigtreetech/BIGTREETECH-SKR-V1.3 #473: SKR 1.3/1.4 and OLED SPI

yet-another-average-joe, 2020-12-03:
Hi,

did someone experiment with such displays : OLED displays using a 4 pin I²C interface. Can't get it to work properly. I have 2 of them : 0.96", 1.3", SSD1306 and SH1106. I also have a 1.53" (SSD1309, 7pin) I still didn't experiment with.

I connect them to the I²C header, and they display what they have to. But after some time :
- random reboots
- vertically shifted display

(I NEVER had the tinyest problem using them with Arduinos and  BluePills and u8g2lib but Marlin uses u8glib, now unsupported)

There's an open issue on Marlin Git, and some users are reporting this behaviour on RepRap forums. I tried each and every solution, but at some point, the issues were back. It seems noboby fixed it, nor found why. Various pullups, wirings, terminations, incantations, voodoo, etc. Believe it or not, the day it worked without an issue was monday 3 (full moon).

https://reprap.org/forum/read.php?13,499572,page=7
https://github.com/MarlinFirmware/Marlin/issues/14431

I received a 1.4 Turbo 5 days ago, and couldn't get rid of these issues. Had a full day without (dec./ 3), the board being powered 24/7, AAnd it finally it came back. I'd like to understand...

yet-another-average-joe, 2020-12-04:
Problem solved.

Slowing down I2C to 100kHz did it ; it was 400 kHz with U8G_I2C_OPT_FAST. Too much.
Frequencies were verified with the scope.

yet-another-average-joe, 2021-08-20:
Forgot about this thread...

There's a mistake in the title : it was about I²C, not SPI

I²C OLEDs do not work with Marlin on these boards because of u8glib ; known issue, on u8glib and Marlin Git ; never fixed. At first slowing down the I²C clock seems to work, but the firmware randomly crashes.

SPI OLEDs work flawlessly (and communicate much faster : 1MHz vs 400kHz).

Domush, 2022-03-04:
> Forgot about this thread...
> 
> There's a mistake in the title : it was about I²C, not SPI
> 

Which pins are you using for i2c? I'm having trouble finding where the actual skr 1.3 board i2c pin names are listed anywhere. The MFG lists 00 and 01, and 27 and 28, but these are clearly not the SKR pin numbers, as 27 is the SD pin and P0_00 and P0_01 are E1 headers which immediately reset the board when I attempt an i2c connection.

yet-another-average-joe, 2022-03-05:
I have a 1.4, not a 1.3.

The I²C pins on the SKR **1.4** are 0.0 and 0.1. I got them working, as many others did (see RepRap forums and Marlin Git). But the display shifts and the board crashes after a few time. u8glib known bug. See u8glib Git.

0.0 = SDA1
0.1 = SCL1

The pins likely are the same on the SKR 1.3 ? Reading the SKR 1.3 schematics, it is **NOT** the case !!!

Reading the SKR **1.3** schematics, it seams there's no spare I²C bus available.

I²C1 (LPC1768 pins 46/47) = E1DIR/E1STEP
alternate I²C1 (LPC1768 pins 58/59) : YSTEP/YDIR
I²C2 (LPC1768 pins 48/49) = E1EN/E0DIR 
SDA0 (LPC1768 pin 25) = P0.27
SCL0 (LPC1768 pin 24) = P0.28

P0.27 is in use for the onboard SD card reader (DAT2) ; probably SD DET
P0.28 is in use : EXP1 pin 9 = encoder button

Conclusion : the SKR 1.3 has no broken out I²C bus, and there's no way to reaffect one.