bigtreetech/BIGTREETECH-SKR-V1.3 #477: SKR 1.4 TURBO part fan cooling not working

polyglotte, 2020-12-08:
Hi guys
I have a issue with this board it keep breaking part cooling fans.
Right now this is the 2nd that is broken (both are from different brands and sellers) in a very short time period, I didn't realize untill I saw bridges bending during printing.
I thought maybe it is because of auto fan from Slic3r 1.3 too heavy duty for a cheap fan (2 to 3 euros from aliexpress).
Then I tried test them in Pronterface the first one looks it still moves at 200 feedrate but can't go over this and it doesn't sound trully 200 feedrate.
The fan is correctly plugged red for + black for -, marlin is set correcly (Previously it worked just fine).
Tried inside Marlin to turn on the part cooling fan, it doens't move at all.
After 2 broken fan now I suspect it is the board problem. What should I do?

jasonfox7, 2020-12-30:
Unless you test the voltage coming out of that fan header with a multimeter there is not much they can do to help you really. It sounds as though you are either delivering too much or too little voltage to the fans and killing them. That would point to a hardware issue with your board.

polyglotte, 2021-01-01:
Is there any solution if it delivers more or less voltage? I'll check voltage.

> Unless you test the voltage coming out of that fan header with a multimeter there is not much they can do to help you really. It sounds as though you are either delivering too much or too little voltage to the fans and killing them. That would point to a hardware issue with your board.

jasonfox7, 2021-01-02:
> Is there any solution if it delivers more or less voltage? I'll check voltage.

Well, yes, but the point is that if either is the case you have a problem and need a new board. That would be better than using a series resistor or diodes to drop voltage, etc. The boards are rather cheap on aliexpress.

Achim01, 2021-01-27:
I had the same problem with the skr 1.4 board.
With a new board, use a FET module at the fan output.
Or use the E1 output when not in use, change the pin configuration for this.
Fan0 output is 2.3 HE1 output is 2.4
Or use the Neopixel port (1.24) with a Fet-Modul, this is a buffered 5v output