bigtreetech/BIGTREETECH-SKR-V1.3 #479: no USB Connection after Marlin Upload - Bricked?

Laube92, 2020-12-18:
Dear all,

after i connected the SKR 1.4 turbo first time with my pc and flashed the Marlin FW the board can not connect to my computer anymore. There will be no device in windows device manager and i can not find it. I tried several SD cards - no success.
I tried to fix the issue by connecting an ST Link V2, but i dont get a connection with the ST Link utility software.
When i click Connect to the target i get the error "select normal mode or HotPlug mode"... i tried all configurations, nothing works to get a connection.

Can anybody help?
Marcel

oldman4U, 2020-12-20:
Why not downgrade to see?

NezSez, 2021-02-20:
Flash the original firmware.bin which came with the stock SDcard back onto the board, assuming the usb connection worked originally with it and that it successfully flashes. If you can connect then you know one of the options/settings you've enabled has caused an issue and you can start to single out the culprits.
It would help if BTT supplied the default configs they used to generate the firmware.bin files they distributed <sigh>.