bigtreetech/BIGTREETECH-SKR-V1.3 #481: Slow booting on SKR V1.3

mrx23dot, 2020-12-22:
There is a 5-6second delay before Marlin's setup( ) is called, so I think it comes from the bootloader. It's also there without SD card inserted.

Is it possible to remove this delay? Why is it there?
Where can I find the source code for the bootloader? Why isn't in this repo?

radek8, 2021-01-10:
https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/issues/346

mrx23dot, 2021-01-10:
https://github.com/Marsman1970/SKR-V1.3-Bootloader 
only contains the original (ripped) bootloader binary, without source code.

@bigtreetech please upload the bootloader source or fix the issue.

radek8, 2021-01-10:
Isn't that enough for you?

https://github.com/ardiehl/BTT_SKR_13_14_14T_SD-DFU-Bootloader

mrx23dot, 2021-01-10:
sorry, I missed that

mrx23dot, 2021-01-10:
continuing here https://github.com/ardiehl/BTT_SKR_13_14_14T_SD-DFU-Bootloader/issues/2

closed 2021-01-10 by mrx23dot