bigtreetech/BIGTREETECH-SKR-V1.3 #483: Endstop Z don’t work on BTT SKR-v1.4 Turbo

Esla2, 2020-12-28:
Hi All,
You are my last chance, I spend days to resolve that.
I put this board into my old 3D( Malyan M180). I wired all the connections. However, I fail on Z Endstop. He don’t want to work.
The others endstop(X, Y) work fine. But Z don’t. He stay ‘Open’ and when I home the Z the bed crash the nozzle.
When I swap connectors Y and Z it’s work. The endstops on X and Y Are Mechanic, Z is not the same.
I ad pic chowing the swap Y and Z ( Y stay open, Z triggered). And Config.h and adv.h
Is there a work around ?
Sorry for my English, It’s not my native language.
And thank you for your help.
[image]
[image]
[config.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/5747907/config.zip)

radek8, 2020-12-31:
https://www.youtube.com/watch?v=EivpEgsfNDc
https://www.youtube.com/watch?v=gdecCW3vsIM

Esla2, 2020-12-31:
Hi Redek8
Thank you for your help.
The two Youtubes, I seen them already. But they didn’t help me. Because my board don’t work even with endstop mechanic.
Now I think my board is defective. I spend a week trying, without result.
Now I’m facing, two possibility: The board is made before february Or the Z channel is defective ( endstop Z and andstop mecanic : Two differents Endstop don't work)
In addition, the Fan0 don’t work too.
I’m going to see that with Aliexpress.

radek8, 2020-12-31:
endstop Z remap to another connector.
You can remap Fan0 to HE1.

Esla2, 2020-12-31:
I use HE1 for a second hotend. I already tryed remap Endstop Z without succes.

Esla2, 2021-01-01:
Hi Redek8,
Happy New Year.
Sorry for my last comment. I was upset about my board.
For remap endstop Z I tried this : https://www.reddit.com/r/BIGTREETECH/comments/ewfqn0/skr_14_turbo_inductive_sensor_not_workinghelp/
Without success. In the post, its talk about switch X and Z but they don’t talk how! If you know how could you please tell how to do it.
Moreover, thank you for your help.

radek8, 2021-01-01:
A small recap.
You are using an optical probe on the Z axis.
does not switch when connected to the Z-MIN connector, but works well when connected to the X-MIN.
Am I correct?

radek8, 2021-01-01:
Connect the optical probe to the X-stop connector.
Connect mechanical endstops to Y-stop and Z-stop
ensure that no endstop is activated and send an M119 command.
Activate all endstops and resend M119.
Write me both results.

Esla2, 2021-01-01:
No, I’m not using optical probe. Just standard endstop. X, Y mechanical endstop, Z optical Endstop, you can see it in my first Pic.
[image]

radek8, 2021-01-01:
> No, I’m not using optical probe. Just standard endstop. X, Y mechanical endstop, Z optical Endstop, you can see it in my first Pic.

OK, I spoke badly.
According to the response M119 t, it seems that the bird stop does not work even after connecting to the X-axis connector.
The optical stop does not seem to work well.

[image]

Esla2, 2021-01-01:
it was my first deduction, but I changed it with this one and it still doesn't work
https://www.amazon.fr/gp/product/B077X8XL56/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1
[image]

radek8, 2021-01-01:
The problem will be in the sensors.
Can you measure the voltage on the signal pin in the activated and deactivated state?

Esla2, 2021-01-01:
With the stock one the value don't change always = 0.01V
With the second one( Amazon) Triggered = 0.36V ; Open = 0.17V

radek8, 2021-01-01:
It is bad.
are you connected correctly?

radek8, 2021-01-01:
the difference between closed and open must be according to the 3.3V logic
high ≥ 2.0
low ≤ 0.8

radek8, 2021-01-01:
connected
V = 5V
G= GND 
S= Signal pin

Esla2, 2021-01-01:
Yes my connection est correct. my connection check twice. and measured 3 times.
V = 5V
G= GND
S= Signal pin  Yes correct.
I have betwine V and G = 4.95V

radek8, 2021-01-01:
The voltage between G and S must be in the range
high ≥ 2.0
low ≤ 0.8
depending on whether it is closed or open.
Is the LED on and off when turned on?

Esla2, 2021-01-01:
The led stay on all the time.

radek8, 2021-01-01:
[image]

Esla2, 2021-01-01:
Yes, mine is always on triggered or not.

radek8, 2021-01-01:
Then you have a problem with the sensor

Esla2, 2021-01-01:
My sensor is like yours. and i bay it from Amazon Three days ago.

Esla2, 2021-01-01:
[image]
[image]
That imply my sensor is faulty. all i have to do is buying others.
thank you very much for your help.
We can close this thread.
Esla2

radek8, 2021-01-01:
Don't buy a new sensor.
Buy 3DTouch.
It's super

Esla2, 2021-01-01:
have you any brand to advice me ?
What i have seen in Config.h it's to tricky to configure !

radek8, 2021-01-01:
https://www.banggood.com/Auto-Bed-Leveling-Sensor-V3_0-3D-Leveling-Touch-For-Reprap-3D-Printer-Part-Mainboard-Heated-Bed-p-1569989.html?rmmds=myorder&cur_warehouse=CN

I bought this and it works great.
I wouldn't want a printer without leveling.
Configuration is not complicated.

radek8, 2021-01-01:
The malfunctioning fan, remap the fan pin and use the external MOFSET module

https://www.aliexpress.com/item/32816837677.html?spm=a2g0s.9042311.0.0.27424c4dutw2hB

Esla2, 2021-01-02:
Thank you for the link to 3DTouch and MOFSET Module.
For remap Fan0, with what pin. I use HE1. They are pin 1.26, 1.25 or 1.27 ( E0DET, E1DET or PWRDET) ?

Esla2, 2021-01-02:
correction 1.29, 1.28, 1.27

radek8, 2021-01-02:
Pins 1.25, 1.26 and 1.0 cannot supply the fan directly, so a module with Mofset is needed.
If it is a fan on the heatsink, or a fan for the cooling of electronics, which is only to be turned on or off, you can use any scented pin. But if you want to regulate the power using PWM control (fiber cooling after extrusion) you have to use a pin that supports PWM control, for example 1.26 (via the Mofset module)
PWM support can be found in the CPU documentation

Esla2, 2021-01-02:
I need fan for my two HE0 and HE1 (#define E0_AUTO_FAN_PIN P2_04; #define E1_AUTO_FAN_PIN P2_04). I understand I need a MOFSET to turn them on ( EXTRUDER_AUTO_FAN_TEMPERATURE  50).
I have Two fans of 12V linked in series. Logically I connect then into FAN0 but FAN0 is faulty.
With MOFSET I have to change the pin in adv.h and in pins_BTT_SKR_common.h
if i'm correct ?

radek8, 2021-01-02:
FAN0 (pin2.3) is predefined for fiber cooling
and is switched by g-code from the printed file.
That's probably why it doesn't sleep at 50 ° C
Use pin 1.25 to cool the heat sink
#define E0_AUTO_FAN_PIN P1_25
#define E1_AUTO_FAN_PIN P1_25

Esla2, 2021-01-02:
All this with the MOFSET that you give the link ?
What is the the g-code command too turn it on ?

radek8, 2021-01-02:
With another pin you can control the switching and power of the driver cooling fan
#define USE_CONTROLLER_FAN
#if ENABLED (USE_CONTROLLER_FAN)
   #define CONTROLLER_FAN_PIN P1_00 // Set a custom pin for the controller fan
     #define CONTROLLERFAN_SPEED_MIN 0 // (0-255) Minimum speed. (If set below this value the fan is turned off.)
   #define CONTROLLERFAN_SPEED_ACTIVE 255 // (0-255) Active speed, used when any motor is enabled
   #define CONTROLLERFAN_SPEED_IDLE 0 // (0-255) Idle speed, used when motors are disabled
   #define CONTROLLERFAN_IDLE_TIME 60 // (seconds) Extra time to keep the fan running after disabling motors
   #define CONTROLLER_FAN_EDITABLE // Enable M710 configurable settings

radek8, 2021-01-02:
https://marlinfw.org/docs/gcode/M106.html

Esla2, 2021-01-02:
OK, I just have to buy MOSFET and 3DTouch
Thank you.

tca72, 2021-03-19:
Hello
On SKR V1.4, and probably on sKRV1.4 turbo there is an hardware bug on optical endstop on Z axis.
There is a wrong resistor on the board ( Z axis). but on other axis it could work.
See videos here:
https://www.youtube.com/watch?v=gdecCW3vsIM
https://www.youtube.com/watch?v=gdecCW3vsIM

zvonka, 2021-05-15:
i had the same problem as you, but with an inductive probe 
it was a configuration.h error, there were some not-obvious changes that had to be made in order for it to work with this board and my printer (tronxy xy2pro)
 
endstop section
// @section homing

// Specify here all the endstop connectors that are connected to any endstop or probe.
// Almost all printers will be using one per axis. Probes will use one or more of the
// extra connectors. Leave undefined any used for non-endstop and non-probe purposes.
#define USE_XMIN_PLUG
#define USE_YMIN_PLUG
//#define USE_ZMIN_PLUG ***********
//#define USE_XMAX_PLUG
//#define USE_YMAX_PLUG
//#define USE_ZMAX_PLUG

// Enable pullup for all endstops to prevent a floating state
#define ENDSTOPPULLUPS. //**********
#if DISABLED(ENDSTOPPULLUPS)
  // Disable ENDSTOPPULLUPS to set pullups individually
  //#define ENDSTOPPULLUP_XMAX
  //#define ENDSTOPPULLUP_YMAX
  //#define ENDSTOPPULLUP_ZMAX
  //#define ENDSTOPPULLUP_XMIN
  //#define ENDSTOPPULLUP_YMIN
  //#define ENDSTOPPULLUP_ZMIN
  //#define ENDSTOPPULLUP_ZMIN_PROBE
#endif

// Enable pulldown for all endstops to prevent a floating state
//#define ENDSTOPPULLDOWNS
#if DISABLED(ENDSTOPPULLDOWNS)
  // Disable ENDSTOPPULLDOWNS to set pulldowns individually
  //#define ENDSTOPPULLDOWN_XMAX
  //#define ENDSTOPPULLDOWN_YMAX
  //#define ENDSTOPPULLDOWN_ZMAX
  //#define ENDSTOPPULLDOWN_XMIN
  //#define ENDSTOPPULLDOWN_YMIN
  //#define ENDSTOPPULLDOWN_ZMIN
  //#define ENDSTOPPULLDOWN_ZMIN_PROBE
#endif

// Mechanical endstop with COM to ground and NC to Signal uses "false" here (most common setup).
#define X_MIN_ENDSTOP_INVERTING true // Set to true to invert the logic of the endstop.
#define Y_MIN_ENDSTOP_INVERTING true // Set to true to invert the logic of the endstop.
#define Z_MIN_ENDSTOP_INVERTING true // Set to true to invert the logic of the endstop.*****
#define X_MAX_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define Y_MAX_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define Z_MAX_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define Z_MIN_PROBE_ENDSTOP_INVERTING true // Set to true to invert the logic of the probe. *****

**probe pin setup***
//#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN //********** remember to leave this disabled if probe is attached to probe pin on board instead of z-min endstop pin
#define USE_PROBE_FOR_Z_HOMING //do enable this though of course
//#define Z_MIN_PROBE_PIN 0_10 // Pin 32 is the RAMPS default
//#define PROBE_MANUALLY
#define FIX_MOUNTED_PROBE //this as well enable

probe offset section:
#define Z_CLEARANCE_DEPLOY_PROBE   2 // Z Clearance for Deploy/Stow
#define Z_CLEARANCE_BETWEEN_PROBES  15// Z Clearance between probe points
#define Z_CLEARANCE_MULTI_PROBE     10 // Z Clearance between multiple probes
#define Z_AFTER_PROBING           10 // Z position after probing is done
#define Z_PROBE_LOW_POINT          -25 // ***** this has to be a number this high or it wont work properly will get probing failed on the first probe if not set to some very high negative number

and homing section:
//#define Z_IDLE_HEIGHT Z_HOME_POS //********
#define Z_HOMING_HEIGHT  4      
#define Z_AFTER_HOMING 10  

#define X_HOME_DIR -1
#define Y_HOME_DIR -1
#define Z_HOME_DIR -1

#define MIN_SOFTWARE_ENDSTOPS
#if ENABLED(MIN_SOFTWARE_ENDSTOPS)
  #define MIN_SOFTWARE_ENDSTOP_X
  #define MIN_SOFTWARE_ENDSTOP_Y
  //#define MIN_SOFTWARE_ENDSTOP_Z //********
#endif

#define MAX_SOFTWARE_ENDSTOPS
#if ENABLED(MAX_SOFTWARE_ENDSTOPS)
  #define MAX_SOFTWARE_ENDSTOP_X
  #define MAX_SOFTWARE_ENDSTOP_Y
  #define MAX_SOFTWARE_ENDSTOP_Z
#endif