bigtreetech/BIGTREETECH-SKR-V1.3 #484: SD card not mounting in Octoprint

parskofamily, 2020-12-28:
Hi,

My SD card is not mounting in Octoprint.  I'm trying to use the Firmware Updater add-on, and the support at that plug-in suggests I need to resolve getting my SD to mount onto Octoprint.

I have gone through the Firmware Updater procedure, but it won't show the card as mounted.

I do not see USB0 when I type the command `df -h` into a Putty terminal.  If my USB was mounted correctly, I should see it there.

Has anyone else come across this that could provide me with some advice?

SETUP:
Octoprint 0.15
SKR1.4 Turbo
Marlin 2.0 bugfix (days old)
(everything is the latest and greatest as this printer is only a few days old, new custom build)

Thanks,
-Parsko

parskofamily, 2021-01-03:
I still haven't figured this out.  Does anyone understand how the SD works?  While I am not able to do this, I would like to add an additional SD card reader (that I have) so I can store and run files from SD.  Is that necessary, or can that be done with the onboard micro SD card which holds the firmware?

I do not have any screen with a built in SD card.  I'm must trying to connect a breakout board to the same pins.

I am pointing out a 2nd issue here, in addition to my initial issue.  I'm primarily trying to solve the issue of updating the firmware.  I think me adding the 2nd SD may help me resolve the first.

Thanks.

parskofamily, 2021-01-17:
Closing due to issue being resolved in link to the Firmware Updater in post 1.

closed 2021-01-17 by parskofamily

galamdring, 2022-01-01:
just came across this, and in case you didn't know, the folders you mount things into in linux exist as normal directories. If you copy something there when the mounted thing isn't mounted, it places the file in the directory on the parent device. When you mount the drive onto that mount point, it hides what otherwise exists in that directory, so you only see what is on the mounted device. When you unmount it, you can see the file that exists on the parent device again.