bigtreetech/BIGTREETECH-SKR-V1.3 #486: Fresh smoothieware

zmrdko, 2020-12-29:
Hi,
i am trying to make fresh smoothieware work with TFT35 v2. The provided binaries here work - the display connects to board.
However if I use fresh binary from Smoothieware's github, everything works except TF35 display. It says: No printer attached!

So my question is, what do I need to change in original smoothieware code to make it work as with bin file from here.
the main reason for this is, that I want to use CNC version of binaries from Smoothieware.

Roughd1433, 2021-01-01:
Just a thought. I know when ruining the BTT tfts on marlin you have to make sure the serial connection speeds for the skr and the tft have to be set the same. IE 115200

zmrdko, 2021-01-01:
Yeah, thats was the first thing I checked.
Anyway, thanks for input

Roughd1433, 2021-01-01:
WHat about the port assignments. I haven't played with smoothieware yet.
It's on the round to it list when I get my cnc router built..

zmrdko, 2021-01-01:
Ports should be fine. RX0 and TX0

radek8, 2021-01-02:
You must have the M503 and M115 commands enabled
Based on them, the display identifies the printer

zmrdko, 2021-01-02:
Thanks for info. I tried both command via console and smoothieware responded. Do you have any other suggestion?

radek8, 2021-01-02:
Did you try to swap the pins in the RX and TX connectors?

radek8, 2021-01-02:
Do you have a correctly defined serial port on pins P0.2 and P0.3 in the smoothieboard?

zmrdko, 2021-01-02:
I tried swapping RX and TX but  no difference. Still No printer attached. Its SKR 1.4 Turbo 
https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/BTT%20SKR%20V1.4/Hardware/BTT%20SKR%20V1.4PIN.pdf
with latest cnc smoothieware
[smoothieware: firmware-cnc.bin](https://github.com/Smoothieware/Smoothieware/blob/edge/FirmwareBin/firmware-cnc.bin)
and this is my config
[config.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/5760897/config.txt)

I dont know where to verify if I have correct settings for serial port 0.

radek8, 2021-01-02:
unfortunately I do not know the configuration smoothieware