bigtreetech/BIGTREETECH-SKR-V1.3 #487: skr 1.4 turbo halts printer at the beginning

Sekai1942, 2021-01-01:
Hi. i have a skr 1.4 turbo with a lcd tft35 v3.
The problem is when i start a new print the bed starts heating after the head starts heating but when it reaches about 237°C or 234°C, the target is 240°C.
It just stops at that point, does not go up at all.
after i restart the board i see that the temperatures are low, it means after the board freezes the value it shows is not real.
And if i try to stop a print from the lcd panel the printer does not go up and to the origin, i have to restart the board.
I do not know what could be the problem.
I have this problem after i updated with latest update from github.

roberttco, 2021-01-11:
you will likely need to recompile the firmware to match your printer - for example, if the firmware thermistor doesn't match yours, the temperature may be incorrect.  When you stop a print, the TFT sends custom GCODE to the printer to move the extruder.  Is your LCD config.ini file set up properly to do this ?  Also you said you have the latest update form Github.  There is two different firmware files - one for the printer and one for the LCD - which do you mean ?

Sekai1942, 2021-01-12:
i mean the firmware for the skr1.4 turbo.
The thing is after i restore from eeprom and save it from the lcd panel.the printer works.
if i shut down the printer and only home the axis and i dont restore  the eeprom it just freezes after i try to print something after some time.
The old firmware had the same setting with the thermistor.