bigtreetech/BIGTREETECH-SKR-V1.3 #490: SKR v1.4 #if _OLD_TEMP_PIN(TEMP_BED_PIN) error compiling

A-Lizard, 2021-01-05:
Can't compile the code to test it on board. The board is SKR v1.4 with lpc1768. 
In platformio.ini:
`default_envs = LPC1768`

In Marlin-bugfix-2.0.6.1-SKR-V1.4\Marlin\src\pins\lpc1768\pins_BTT_SKR_V1_4.h:
`#define TEMP_1_PIN                      P0_23_A0  // A2 (T2) - (69) - TEMP_1_PIN`
`#define TEMP_BED_PIN                    P0_25_A2  // A0 (T0) - (67) - TEMP_BED_PIN`

In Marlin-bugfix-2.0.6.1-SKR-V1.4\Marlin\src\HAL\LPC1768\inc\SanityCheck.h:
`#define _OLD_TEMP_PIN(P) PIN_EXISTS(P) && _CAT(P,_PIN) <= 7 && _CAT(P,_PIN) != 2 && _CAT(P,_PIN) != 3`
`#if _OLD_TEMP_PIN(TEMP_BED_PIN)`
` #error "TEMP_BED_PIN must be defined using the Pn_nn or Pn_nn_An format. (See the included pins files)."`
`#elif _OLD_TEMP_PIN(TEMP_0_PIN)`
`  #error "TEMP_0_PIN must be defined using the Pn_nn or Pn_nn_An format. (See the included pins files)."`
and so on ...

The error that terminal is throwing in SanityCheck.h: 
[error operator defined](https://pastebin.com/ASKrE60n)

I use Configuration and Configuration_adv files from the BTT SKR V1.4 Marlin 2.0.6.1 archive (although my printer is A8 plus). Can you help me ?

oldman4U, 2021-01-07:
Use a Marlin firmware.

Google how to "reset" VSC.

A-Lizard, 2021-01-08:
They are reset, but even after restart of the program, compile error remain.

oldman4U, 2021-01-11:
Problem solved?

Please help the community and close the ticket once you do not need it anymore. Thank you

closed 2021-01-12 by A-Lizard