bigtreetech/BIGTREETECH-SKR-V1.3 #494: just cannot use the capacitive proximity sensor with SKR 1.4 Turbo

saintofinternet, 2021-01-21:
the M119 reports either TRIGGERED or OPEN..

if it is OPEN and gets TRIGGERED it stays in the Triggered mode.

else it just stays in OPEN MODE.

the sensor i use is the https://in.element14.com/autonics/cr18-8dp/proximity-sensor-capacitive-8mm/dp/1736831

it was working perfectly fine with the Arduino+RAMPS board but now with the SKR 1.4 Turbo board it's giving me  a lot of issues.

radek8, 2021-01-31:
Some users complain of a problem with the Z-stop connector when using the probe.
As a solution, they state the confusion with the Y-stop connector
with the appropriate modification in the pinout file

FunnyGermanGuy, 2021-03-08:
Which stepper drivers are you using? If its a TMC maybe you forgot to bend or cut the pin for stallguard on the z axis stepper driver? Because if so the endstop input is unfunctional. If you wire the sensor some funny ways you can read some inputs from the sensor but it wont work until you unconnect the stallguard pin on the driver