bigtreetech/BIGTREETECH-SKR-V1.3 #496: SKR 1.4 Turbo (TMC 2209) no connection to TFT2.4 V1.1

lullaby37, 2021-01-30:
Hey folks,

as my SKR 1.3 dies recently, I upgrade to a 1.4 Turbo. I wanted to use the TFT24 screen I had previously connected to my 1.3 and iot does work in Marlin mode.
When I switch to TFT touch mode I only get "No printer attached".
The strange thing is that I cannot read values like temperature or fan speed but I can, for example, set the fan speed to 100% using the touch screen.

What I did so far
1) Checked wiring, swapped RX and TX and back.
2) upgrade TFT with latest firmware from github
3) Changed serial port in MArlin Code but it seems on ly -1 and 0 compile. When switching to 2 or 1 I get several errors
4) Googled for answer but nothing helped. I read one thread where they changed somwthing in the TFT-Firmware but I did not understand what and how.

Hoping there is someone out there who can help me.

radek8, 2021-01-31:
send configuration files from both Marlin and TFT

lullaby37, 2021-02-02:
Hi radek,
thanks for your reply.
I used the latest configuration from BTT github for the TFT.
As for the Marlin files, I attached the configuration and configuration advance.
Thanks for your help, man!

[Configuration_adv.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/5909906/Configuration_adv.txt)

[Configuration.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/5909894/Configuration.txt)

radek8, 2021-02-02:
You have to allow
#define M114_DETAIL
#define REPORT_FAN_CHANGE
#define M115_GEOMETRY_REPORT

But that probably won't be the main problem.
How did the SKRV1.3 board die to you?
Couldn't one pin on the SR232 connector on the TFT be blown?

radek8, 2021-02-02:
Try to modify the configuration as I wrote you.
If that doesn't help, we'll test the RX and TX pins for defects

lullaby37, 2021-02-03:
Hi radek,

I changed the settings but still there is no change. the printer does not connect to the tft.

The SKR1.3 I used before died due to a broken stepper cable. The extruder did stop working for some reason.  I swapped cables and stepper drivers to see where the problem came from when suddenly the "TMC driver error" was in the display even though no driver was plugged in. Also the chip got extremely hot.
Actually I am 90% sure that I did not put the TMC driver incorectly but it could be that a pin got connected incorrectly... That may have cause the board to burn. It does boot up but it gets hot and the error still is in the display.

radek8, 2021-02-03:
Can you check that the RS232 cable is 100% OK?
If so, the pins on the TFT display may have been damaged.
If the SKR V1.4 board is new, then hopefully it will be fine.

lullaby37, 2021-02-03:
Hi radek,
I can now definitely say that it is not a hardware but a software issue.
I connected the TFT24 to my old broken SKR13 and after some seconds it connects to this board and I can read temperature values and so on. So the cable works and the pins are not burnt.

I read somewhere that changing USART-ports could help?
I have no idea how to do this I have to say.

lullaby37, 2021-02-03:
I am an idiot... haha...
For testing it on the SKR13 I had to remove the cable. It seems like the cable was bent too much and somehow disconnected.
Now I have conneted everything back to the SKR14 it now it does work.
Thanks for giving the hint for the cable. That probably was the error

radek8, 2021-02-03:
Great. You found a defect

closed 2021-02-04 by lullaby37