bigtreetech/BIGTREETECH-SKR-V1.3 #498: SKR V1.4 upload firmware Marlin 2.0.7.2

killrob, 2021-02-03:
My SKR V1.4 with TMC2209 and TFT35 E3 V3.0 don't load the firmware.bin file compiled with Vscode and Platformio.
If i check the microsd again in the PC the file is still .bin  and not .cur as expected.
When i power on the printer with the micro SD inserted in the MB slot, the LCD display "printer not connected" for a very long time (about 5 mins) then make connection, but the firmware was not uploaded and also the Marlin mode is not working.

oldman4U, 2021-02-04:
Try another sd card smaller 8GB FAT32 formatted. Remove everything attached to the mainboard except power.

killrob, 2021-02-04:
> 
> 
> Try another sd card smaller 8GB FAT32 formatted. Remove everything attached to the mainboard except power.

ok i will try, and... you know , in any case, if there is a way to reload the bootloader with Arduino?

oldman4U, 2021-02-04:
What makes you believe the bootloader is broken? Start with the most likely
and not with the last likely.

killrob, 2021-02-04:
> 
> What makes you believe the bootloader is broken? Start with the most likely and not with the last likely. 
just because i've done this steps.
switch the jumper to USB power, used a 128MB sd card FAT32, connected to PC. And the PC don't show me the content of the SD.
I've done same steps on another SKR 1.4 i have and the PC shows me the files in the SD

oldman4U, 2021-02-04:
Somehow important this information. Contact your dealer for a replacement.

killrob, 2021-02-04:
hmmm ok, but if there is a way to correct this with arduino will be better, i'm always ready to learn something new :)

oldman4U, 2021-02-04:
Sorry but this I do not know.

radek8, 2021-02-06:
look here

https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/issues/346